Klipper3d/klipper #2420: Flymaker Flyboard Connection Problem

AniInABox, 2020-01-23:
Hello,

i have bought the Flymaker Flyboard with the STM32F407 CPU.
I managed to get the Klipper firmware flashed to the Board.

Problem is the USB Connection.
Log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4102633/klippy.log)

Config:
[Klipper Flyboard.txt](https://github.com/KevinOConnor/klipper/files/4102625/Klipper.Flyboard.txt)

AniInABox, 2020-01-24:
I read the other Issue for the Config.
I changed to this und got a new Errors:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4107246/klippy.log)

KevinOConnor, 2020-01-24:
Your logs indicate the micro-controller code is not running correctly.  Try pulling the latest code (as described in the FAQ), run `make distclean`, and then complete all the board flashing steps (starting with `make menuconfig`).

This is also a very new board and I don't know if anyone has had success with it.  (Perhaps @DJDOUBLEG has?)

-Kevin

AniInABox, 2020-01-24:
Ok, i have done all this steps, same error.
In the Repository of flymaker there is a Marlin Version for this Board.
In the Configuration.h he set the Serial to 1 (not 0 which is normal for all other Boards)

Can this help to find the error?
Can i do something more to get more Information?

DJDOUBLEG, 2020-01-24:
> Your logs indicate the micro-controller code is not running correctly. Try pulling the latest code (as described in the FAQ), run `make distclean`, and then complete all the board flashing steps [...]

Under testing.
I don't enough time at now.
Only have time to make pinout config for klipper.

DJDOUBLEG, 2020-01-28:
> 
> 
> Ok, i have done all this steps, same error.
> In the Repository of flymaker there is a Marlin Version for this Board.
> In the Configuration.h he set the Serial to 1 (not 0 which is normal for all other Boards)
> 
> Can this help to find the error?
> Can i do something more to get more Information?

Hi AnilnBox!

I have same issue.

[image]

AniInABox, 2020-01-29:
Hi, welcome to the club.
Do you have some clue why this happens?
I am not familiar with Micro Controller programming.

Changes to Klipper: Maybe, when i learn Python.
But Micro Controller are not my favorite.

I can test anything, provice debug Information etc. when someone tell my how.

KevinOConnor, 2020-01-29:
The logs in this issue indicate that the Klipper micro-controller code is successfully running on the target board to the point that USB negotiation completes and the initial communication handshake completes.  The micro-controller clock also appears to be functioning correctly and at the expected rate.  However, the code fails during the configuration phase.  If I had to guess, I'd say a likely culprit is the bootloader leaving some device on the micro-controller enabled when entering into the Klipper code and that is causing some memory corruption.

To track down the problem, it generally requires a developer with the board in hand along with a low-level hardware debugger attached to that board.  The low-level debugger can then, ideally, be used to figure out what the source of the corruption is.

-Kevin

AniInABox, 2020-02-27:
Here is an Answer from the Developer from the Board:

klipper FW requires DVECT_TAB_OFFSET = 0x8000
  The bootloader writes the klipper FW to the address after 0x8008000

KevinOConnor, 2020-02-27:
Setting a "32KiB bootloader" in "make menuconfig" sets the firmware address to 0x8008000.  So, the problem must be something else.

-Kevin

kakou-fr, 2020-03-21:
it work for me

DJDOUBLEG, 2020-03-21:
Ooooo this is good new!

kakou-fr <notifications@github.com> ezt írta (időpont: 2020. márc. 21., Szo
23:40):

AniInABox, 2020-03-24:
Yeah i get an USB connection. 1 Time. But this is not the fault of Klipper. So i close this ticket.

closed 2020-03-24 by AniInABox