Klipper3d/klipper #2421: What is line #3 on the right hand side of the display?

dmitrif, 2020-01-23:
Hi! There seems to be a bit of confusion as to what line 3 on the right hand side is:

[image]
[Reddit thread by u/unsoluble](https://www.reddit.com/r/klippers/comments/es3223/what_time_is_being_displayed_on_the_lcd/)

I took a quick look at the code, and there does not seem to be a relevant output, elapsed / remaining time are displayed on the bottom line (that switches between messages).

https://github.com/KevinOConnor/klipper/blob/0a86190bae46a22a493f447075ab7713fd6ae9ed/klippy/extras/display/display.py#L140

JohnEdwa, 2020-01-23:
AFAIK the displays are generated by the menu system using menu.cfg, and I think the RepRap uses [the 1604 layout here](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/display/menu.cfg#L707). So while printing from octoprint and not USB or SD card, that would be `__card_prt_time_2` which is simply formatting what `toolhead.printing_time` returns.

dmitrif, 2020-01-23:
Thank you, John!

closed 2020-01-23 by dmitrif