Klipper3d/klipper #2422: [FEATURE REQUEST] Delta Out of bounds printing

casedog21, 2020-01-23:
This isnt a must have but, this would be a super awesome feature.
I wish I was smart enough to contribute code to this but the fact of the matter is I am not.

Feature request for delta printers (possible for cartesians) - 

**Simpler version**
- Allow config flag true/false to be set
- Allow to print outside of the bed build volume. 
- Rely on the end user to orient the part properly to not smash into the towers with the extuder

**Safer Version**

- tell klipper firmware where the vertical posts are as a "no-go" point.
- IE Manually lower the printer down 10 MM from the bed and almost touching the center point one of the tower for each tower
- Then add in the post width
- Maybe repeat for the upper most point of each tower to set a boundry line. 
- Allow for printing outside of the build volume.

**Usage/Reason for needing it**
I print a lot of large items that I have to split since I have switched to Klipper. I have been using this firmware for a long awhile and advocating for it. I have a Tevo Little Monster. I print RC airplane parts a lot on my printer. With the smoothieware I could orientate a wing part so as it printed the tip of the wing could hang outside of the bed a little. Since since I switched to Klipper, this is causing me to have to split my parts and print multiple pieces instead of one solid piece like I used to. This makes it to where I have to rely on a glue joint as a week point vs the molten filament when I was printing them whole. With Klipper, it errors out because it is essentially out of bounds of the build volume. 

Thanks for reading!!

KevinOConnor, 2020-01-23:
Thanks.  It would help if you could provide a klipper log file (or at least the printer.cfg file) so that we have more information on your printer.  Also, what are the dimensions of the parts are you hoping to print?

-Kevin

casedog21, 2020-01-23:
My print bed is 340x500mm
I would like to be able to extend out at another 50 or more. 
When I do the delta calibrate method manual It doesnt seem to reach out as far as I am used to. Which is probably a config problem. However, I am asking to extend out beyond that. 

I will get the config when I get home. 
Thanks!

-Casey

casedog21, 2020-01-24:
See attached

[klippylog.txt](https://github.com/KevinOConnor/klipper/files/4106418/klippylog.txt)
[printercfg.txt](https://github.com/KevinOConnor/klipper/files/4106419/printercfg.txt)

casedog21, 2020-01-24:
This is an example of the kind of printing I would like to do. This is a buddy of mines work but you can see with smoothieware where you can print outside of the bed. 

[image]

KevinOConnor, 2020-01-24:
Okay.  I added a print_radius option that will allow one to customize the valid XY coordinates (commit 2ea36312).  Note though, if you increase this value, it becomes possible to command the toolhead into a tower.

It's also possible to add additional checking to avoid tower collisions, but an interested developer would need to code and submit it.

-Kevin

casedog21, 2020-01-24:
@KevinOConnor 
Thank you so much! I will give it a shot this weekend! 
Smoothieware never checked for tower collisions. It was up to the end user
to be responsible enough to avoid the towers by part placement. 
I really appreciate this. It is a game changer / time saver for my printer. 

-Casey

casedog21, 2020-01-24:
Is the print_radius in addition to delta_radius or does it override delta_radius?
So if I this to 10 it would allow 10mm in extra print radius or is it to be set print_radius = desired overprint + actual delta radius?

-Casey

KevinOConnor, 2020-01-24:
> Is the print_radius in addition to delta_radius or does it override delta_radius?

The print_radius is an absolute value.  So, if you set `print_radius=10` then a `G1 X15` move would report an error.

If you're looking to fully utilize a 500mm bed, you would set this to `print_radius=250`.

-Kevin

KevinOConnor, 2020-01-24:
Also, it's worth noting that there are still restrictions imposed on the build radius (for example, you still can not command the toolhead to a position further than an arm length from a tower).  Upon starting the software, you should find a couple of lines in the /tmp/klippy.log file that look like:
```
Delta max build height 510.09mm (radius tapered above 478.13mm)
Delta max build radius 156.11mm (moves slowed past 219.47mm and 234.39mm)
```

You should check that your print fits in the area described by the above lines from your log file.

-Kevin

casedog21, 2020-01-24:
Kevin, 

Thanks for the explanation!

So those values that you seeing are the calculated vales after running the delta calibration. 
Maybe its an education thing but I havent been able to fully reach the bed radius that I manually entered. I tried bumping up delta_radius before a calibration before and it seems to get closer to what I am wanting for my bed radius. After Calibration though it always shortens the build radius. You can see in that log that it is initially set to delta_radius and after calibration that gets shrunk down to 156.11. So no my overall diameter gets shorted by 8mm. I havent quite figured out the magic sauce for this yet. 

Example would be, if the radius of the print was 140, it would error out even though the delta_radius  shows that it can go up to 156.11. 

The restrictions make sense.

I can run some tests this weekend and get some more logs if you are interested in seeing them. I can do a test with and without print_radius.I believe your code change will fix this issue altogether though. 

Thanks!
-Casey

KevinOConnor, 2020-01-24:
> Maybe its an education thing but I havent been able to fully reach the bed radius that I manually entered. I tried bumping up delta_radius before a calibration before and it seems to get closer to what I am wanting for my bed radius.

The `delta_radius` parameter is not related to bed radius - it's value depends on the location and characteristics of the towers, tower carriages, and effector.  It would not make sense to alter `delta_radius` to change the available "build envelope".  Use the `delta_radius` that the DELTA_CALIBRATE command calculates.

The new `print_radius` parameter can be used to alter the available "build envelope".  (If you change the `print_radius` configuration, you should see the "Delta max build radius" reported in the log also changes.)

> I can run some tests this weekend and get some more logs if you are interested in seeing them.

Great, thanks.

-Kevin

casedog21, 2020-01-27:
I ran some tests and it worked as expected.   I had some issues with my
printer but I was able to manually move the print head out to the limits
without error. Then once I breached that limit it through the error as
expected.

Thanks again!

-Casey

KevinOConnor, 2020-02-14:
Okay, I'm going to close this as it appears commit 2ea3631 addresses the original issue.

-Kevin

closed 2020-02-14 by KevinOConnor