Klipper3d/klipper #2424: Software SPI fails on more than 4 steppers

BoHoehne, 2020-01-23:
I tried to configure 5x tmc5160 with software SPI. 
The board is a re-arm with a RAMPS 1.6+ on top.

I prepared and tested a wire loom for 5 tmc5160.
The loom was end to end tested with ohm meter (all ok).

I configured one stepper a time.
Restarted klipper service.
Tested with dump_tmc the stepper (if in doubt, did a init_tmc before)

Added another.
Restarted klipper service.
Tested...
and so on...

After adding the fith, apparently, the SPI bus collapsed.
None of the steppers where accessible anymore.

The hardware SPI on the RE-ARM is unusable.

[klippy_log.zip](https://github.com/KevinOConnor/klipper/files/4105180/klippy_log.zip)
[printer_cfg.zip](https://github.com/KevinOConnor/klipper/files/4105181/printer_cfg.zip)

jalanjarosz, 2020-01-24:
I have multiple TMC-2130's on a prototype RAMPS-32 board on a Re-ARM, however I'm using the built-in SPI.

I don't have TMC5160's and I don't think I have enough free TMC2130's to fill the board.

My TMC section looks like this:
[tmc2130 stepper_x]
cs_pin: P1.1 #J12_12 #J12 Header
microsteps: 16
interpolate: True
run_current: 0.87 #Don't go below 0.75 
hold_current: 0.75
sense_resistor: 0.110
stealthchop_threshold: 0

We have the board design much like the BTT RAMPS 1.6+ however we are making a specialized RRD adapter to pull up teh J3, J5, and J12 to use the onboard SPI and Cs of the RAMPS.

jalanjarosz, 2020-01-24:
Looking at the Re-ARM schematic, you should be able to use the BTT RAMPS 1.6+ built in MOSI, MISO and SCK pins with jumpering the Cs pins from the J12 header.

BoHoehne, 2020-01-24:
I switched to software SPI after i discovered that the hardware SPI  (issue #2416) did not work on my 5x tmc5160. 
At the time i issued the #2416 i was not aware of the max. 4 stepper limitation.

I guess the limitation to max. 4 stepper drivers addressable via software SPI is also the case with hardware SPI.  Thats why i wasn't able to get hardware SPI running.

Looking forward for a more improved version of RAMPS.

KevinOConnor, 2020-01-24:
Klipper has no software limitations on the number of spi devices.  Your description sounds like a wiring issue (perhaps the wiring to the 5th device is causing the miso pin to be unintentionally grounded, or perhaps the 5th driver is defective and causing the miso pin to be grounded, or perhaps the 5th cs pin is not wired correctly causing that driver to pull miso low).

-Kevin

BoHoehne, 2020-01-26:
Ok, took a while to get to testing, again.

The cable loom came with the stepper drivers. 
As it turns out, the last connector for miso had a loose connection (on/off/high resistance).

Never trust cables which you haven't tested yourself.

Please accept my apologies.

closed 2020-01-26 by BoHoehne