Klipper3d/klipper #2425: Klipper suddenly wont connect after a forced disconnect due to the bed heating too slow

iH8teF1ames, 2020-01-25:
after the crash it will only give me this error when connecting. I didnt change any settings after the disconnect.

Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready

[klippy.log](https://github.com/KevinOConnor/klipper/files/4111583/klippy.log)

BlackStump, 2020-01-25:
[mcu]
serial = /tmp/printer   <======== that is not correct in your printer.cfg 

it should be something like this
>[mcu]
>serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 

have a read it will help you find the right id
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

iH8teF1ames, 2020-01-25:
> [mcu]
> serial = /tmp/printer <======== that is not correct in your printer.cfg
>
> it should be something like this
>
>
> have a read it will help you find the right id [...]

thats how it was, I changed it while messing with it a few minutes ago just to see if it would change anything. Ill change it back and see if anything has changed.

BlackStump, 2020-01-25:
are you running a pi? and you ran
> ls /dev/serial/by-id/* from an ssh terminal on the host machine?
If it found nothing then either your usb cable is bad or you have problems on the mcu
without knowing what mcu you have you could try reflashing your firmware

iH8teF1ames, 2020-01-25:
> are you running a pi? and you ran
> 
> > ls /dev/serial/by-id/* from an ssh terminal on the host machine?
> 
> If it found nothing then either your usb cable is bad or you have problems on the mcu
> without knowing what mcu you have you could try reflashing your firmware

Yes I'm running it from a pi. 
I ran /dev/serial and a ssh terminal with no response. I just tried a different pi with a different cable and it still wouldn't tell me the serial port so I'm thinking this is an issue with the board (skr mini e3)

cmock13, 2020-01-25:
I am experiencing a very similar issue. Printer was running last night. Restarted this evening and Klipper give me the same error. The klippy.log file states:

> Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD5055252323026610743-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD5055252323026610743-if00'

[klippy.log](https://github.com/KevinOConnor/klipper/files/4111635/klippy.log)

iH8teF1ames, 2020-01-25:
>I am experiencing a very similar issue. Printer was running last night. Restarted this evening and Klipper give me the same error. The klippy.log file states: ...

@cmock13 what board are you running? SKR?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

cmock13, 2020-01-25:
I am running the SKR Mini E3 V1.2. I fixed my issue by reloading the firmware.bin file. 

Now I wonder what happened and how I can prevent it again.

iH8teF1ames, 2020-01-25:
>I am running the SKR Mini E3 V1.2. I fixed my issue by reloading the firmware.bin file.  ...

@cmock13  could you link your firmware.bin? I tried to use my old one but it wouldn’t flash on my board. Tried a different SKR and it flashed so I’m hoping it’s just a fluke but it’s looking like a replacement will be necessary 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

cmock13, 2020-01-25:
Yes,

[https://drive.google.com/file/d/1xF8hdh6y4htHHEgr6u8WHBt4e0AwNZA6/view?usp=sharing](https://drive.google.com/file/d/1xF8hdh6y4htHHEgr6u8WHBt4e0AwNZA6/view?usp=sharing )

Good luck!

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like the original report was tracked down to a hardware issue.

-Kevin

closed 2020-02-14 by KevinOConnor