Klipper3d/klipper #2426: [FEATURE REQUEST] low cost hall based Filament Width Sensor

test3210-d, 2020-01-25:
My low cost hall based Filament Width Sensor used two sensors ss49e  and use two adc channel. 
Sensors work in differential mode and temerature drift minimal.  
Now all work on arduino nano, but i think klipper can calculate diameter itself.

https://3dua.info/topic/21-datchik-diametra-filamenta/?tab=comments#comment-156

Sensor use 2 calibration point im my case 
Dia1=1.50 and Dia2=2.00 mm (i use drill rod)

Raw sensor value = adc 1 value + adc 2 value , i use sum from 6 measurements

diameter = (Dia2 - Dia1)/(raw_Dia2-raw_Dia1)*(current_raw-raw_Dia1)+Dia1

[sensor config]
Dia1: 1.50 # (mm)
Dia2: 2.02 # (mm)
raw_Dia1:  4000 #raw 1/5mm
raw_Dia2: 5000 #raw 2 mm
pin1: analog11
pin:2 analog12

# and common  from
# TSLl401CL Based Filament Width Sensor 

default_nominal_filament_diameter: 1.75 # (mm)
max_difference: 0.2
measurement_delay: 100

all commands replicated from tsl401 and one own command for read raw value for calibration

test3210-d, 2020-01-25:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4112242/klippy.log)

KevinOConnor, 2020-01-28:
Thanks.  It is certainly possible to run Klipper directly on the micro-controller and have it directly read the ADC values.  It should also be possible to do that with just Python code (and no C code).  However, it is very unlikely a developer will be able to do that without the hardware in hand - so I suspect someone with the device and experience with software development would be needed in order to implement this.

-Kevin

test3210-d, 2020-01-28:
Thanks for the answer. Unfortunately, I do not have enough experience with python. It is not very difficult for me to write a program for arduino, but I don’t know how to create a separate module for the sensor and the necessary parameters for it in python. However, having a template I can change it if necessary. If it is possible to create a template that reads two ADC channels, then it will be tested and edited if necessary.

KevinOConnor, 2020-01-30:
FWIW, I can point you to some existing code that does related work:
* MCU_ADC_buttons in klippy/extras/buttons.py
* klippy/extras/tsl1401cl_filament_width_sensor.py

Also, be sure to read through the guide at: https://www.klipper3d.org/Code_Overview.html#adding-a-host-module

-Kevin

test3210-d, 2020-01-31:
Thanks ! I do module. It tested and work fine. 
Not inplemented - virtual filament runout function if width out of tolerance, i think function may be implemented in my module and in tsl1401cl module to. 

[hall_filament_width_sensor.zip](https://github.com/KevinOConnor/klipper/files/4139152/hall_filament_width_sensor.zip)

test3210-d, 2020-02-01:
Calibration procedure
1 step add to config 

[hall_filament_width_sensor]
adc1: analog11 #adc channel 1 select own chanel
adc2: analog12 #adc channel 2 select own chanel
Cal_dia1: 1.50 #REFERENCE DIAMETER 1 (mm) select own diameter 
Cal_dia2: 2.00 #REFERENCE DIAMETER 2 (mm) select own diameter
Raw_dia1:9630  #RAW VALUE FOR REFERENCE DIAMETER 1
Raw_dia2:8300  #RAW VALUE FOR REFERENCE DIAMETER 2
default_nominal_filament_diameter: 1.75 # This parameter is in millimeters (mm)
max_difference: 0.15 # mm
measurement_delay: 70 #mm  select own distance 

insert calibration rod (reference diameter 1) and send QUERY_RAW_FILAMENT_WIDTH command. Get raw sensor value for Raw_dia1 parameter
insert second calibration rod (reference diameter 2) and send QUERY_RAW_FILAMENT_WIDTH command. Get raw sensor value for Raw_dia2 parameter
Store parameter in config. All done. 
new design for sensor https://www.thingiverse.com/thing:4138933

KevinOConnor, 2020-02-02:
Thanks.  If you're interested in contributing this to the main Klipper repo then please create a github pull request - see: https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md

-Kevin

closed 2020-02-02 by test3210-d

puneit-hassio, 2020-02-06:
@test3210-d where are the adc pins ? What do you mean by host — is it raspberry pi or main board of the printer. ?

komandrik, 2020-02-06:
@puneit-hassio Printer board

test3210-d, 2020-02-06:
today i record video about sensor 
https://youtu.be/TDO9tME8vp4
i use aux2 connector on ramps-like board

alocin73, 2020-05-02:
Hi @test3210-d ,

thanks for your work.

I've done the hall sensor and it works well.

I've found a problem configuring the menu.
After copying from the doc i found some missing space.
Without spaces I receive a parser error.

After items:

`[menu __filament]
type: list
name: Filament
items:
  __temp __hotend0_current, __temp __hotend0_target
    .__unload
    .__load
    .__feed
  __filament_width_current
  __filament_raw_width_current`

Thanks again,

Nicola

sedlacekdavid, 2020-10-22:
Hi, I like the idea. How about accuracy? Did anyone test it?

Bob90, 2021-02-26:
Can we link to the config page to make this easier to search on google. I have a hard enough time searching for it and I've read it already.
https://github.com/KevinOConnor/klipper/blob/master/docs/HallFilamentWidthSensor.md