Klipper3d/klipper #2427: Klipper freezes after activating Manual Mesh Bed Leveling

Gamesnic, 2020-01-25:
I just tried to activate Manual Mesh Bed Leveling on my Ender 3 running Klipper with a Raspberry PI 3B. Also there is a Version 1.1.5 Silent Board with TMC 2208 drivers in there, running perfectly fine with klipper, which is weird.
Even though DWC2 as Web Interface is able to start and is reachable, the Ender 3 won't respond. The display is frozen, and when I flip the power switch up and down, the display will be blank. I also attached the klippy.log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4112417/klippy.log)

theopensourcerer, 2020-01-25:
In your klippy.log:

"NoOptionError: No option 'mesh_min' in section: 'bed_mesh'"

theopensourcerer, 2020-01-25:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L60

Gamesnic, 2020-01-25:
Tried to change that from min_point and max_point to mesh_min and mesh_max. Still no change.
Here's the updated klippy.log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4112476/klippy.log)

theopensourcerer, 2020-01-25:
"Error: bed_mesh: invalid min/max points" 

As a guess, try removing the second 0 from your coordinates.

Gamesnic, 2020-01-25:
With the comma fully removed, I get this error regarded to mesh_min:
`Error: bed_mesh: malformed 'mesh_min' value: 10`
With the second 0 removed, leaving the first 0 with the comma, I get this error again:
`Error: bed_mesh: invalid min/max points`

theopensourcerer, 2020-01-25:
"#mesh_min:
#   Defines the minimum x,y coodinate of the mesh for rectangular beds.  This
#   coordinate is relative to the probe's location. This will be the first
#   point probed, nearest to the origin. This parameter must be provided for
#   rectangular beds."

Your end stop position is 0. I guess this probably means that you can't have it at zero. i.e. because you might not actually be "on-the-bed"

Gamesnic, 2020-01-25:
As I changed the x and y endstops to 1, I still get the same error with invalid points.

theopensourcerer, 2020-01-25:
What? I think you need to have the mesh min and max INSIDE the dimensions of your bed... 

So for example, if your endstop positions are 0 and your bed is 200 in X and Y then make your mesh start at say 10,10 and end at 190,190.

Gamesnic, 2020-01-25:
Oh, really sorry about that. I misunderstood you completely.
Changing that Fixed everything, thank you for your help.

closed 2020-01-25 by Gamesnic

theopensourcerer, 2020-01-26:
Glad it is sorted. 

@KevinOConnor Perhaps the documentation could be made a little clearer?