Klipper3d/klipper #2429: CI with github actions and docker

luqasz, 2020-01-26:
Hi.

I was wondering if I can help you setup and use github actions (which is free for open source) so you can run tests etc. directly on docker. Travis main disadvantage is that you can not run tests within your customized docker image. I've setup some workflows for my projects. Also one benefit is that you have same docker image which you can test localy on. Check them out and I will be happy to help:

https://github.com/luqasz/avr/blob/master/.github/workflows/ci.yml
https://github.com/luqasz/avr/actions

https://github.com/devpkg-com/gcc-avr/blob/master/.github/workflows/ci.yml
https://github.com/devpkg-com/gcc-avr/actions

KevinOConnor, 2020-01-28:
FYI, I don't have any knowledge of "github actions", so can't comment on the advantages and disadvantages of converting.  As far as I know, travis works for what it does.  Is there some specific feature you'd like to see implemented that can't be implemented today?

-Kevin

luqasz, 2020-01-28:
What does github actions provide:
* Build in parallel with matrix (e.g. build for multiple MCUs at same time)
* Resulting builds may produce artifacts (which can be used as downloadable flash images)
* Build on docker. Bundle all necessary requirements (compilers etc.) in docker, no need to install them all over again. You can use same docker image on CI and developer machines.
* Split script `travis-build` into multiple steps. When one fails, rest won't be executed.
* Cache files if any are required.

IMO main advantages are:
* docker
* matrix build
* artifacts

KevinOConnor, 2020-01-30:
I guess you are proposing running the existing Klipper build checks with a different process.  If that is so, then my high-level feedback is that it's not something I, personally, am looking to spend time on right now.  The travis system works for what it is.

-Kevin

closed 2020-01-30 by luqasz