Klipper3d/klipper #2433: Issue with accessing virtual serial ports

Delsian, 2020-01-26:
Klippy doesn't play well with a second virtual port. This is known issue in PySerial:
https://github.com/pyserial/pyserial/issues/59

```
Unhandled exception during post run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 183, in run
    m.microcontroller_restart()
  File "/home/pi/klipper/klippy/mcu.py", line 676, in microcontroller_restart
    self._restart_arduino()
  File "/home/pi/klipper/klippy/mcu.py", line 644, in _restart_arduino
    serialhdl.arduino_reset(self._serialport, self._reactor)
  File "/home/pi/klipper/klippy/serialhdl.py", line 253, in arduino_reset
    ser.dtr = True
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/serial/serialutil.py", line 469, in dtr
    self._update_dtr_state()
  File "/home/pi/klippy-env/local/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 22] Invalid argument
```

Unfortunately, proposed variant with setting rtscts=True and dsrdtr=True doesn't help. Will continue investigations.

KevinOConnor, 2020-01-28:
Klipper uses pyserial v3.4 (see `scripts/klippy-requirements.txt`) which does not have the above issue, so I'm not sure what is being reported here.

-Kevin

Delsian, 2020-02-01:
Resolved by setting for virtual ports:
"restart_method: command"

closed 2020-02-01 by Delsian