Klipper3d/klipper #2434: Novice with bltouch and klipper

foncubierta, 2020-01-26:
Good, I just installed klipper and bltouch following the different internet tutorials but I get errors in which I can't find a solution.
Normally the nozzle crashes into the glass of the bed and I have to turn off the printer. On other occasions he tells me this:
Recv: // Failed to verify BLTouch probe is raised; retrying. Recv: !! BLTouch failed to deploy Recv: !! BLTouch failed to deploy
On another occasion I ran a BED_MESH_CALIBRATE and I was able to check in the octoprint terminal the different measurements I was giving although I did not end up indicating that I indicated an error.
I sincerely ask someone to explain to me how to start from scratch and I am doing it step by step since I have not found the solution for 2 weeks. Thank you.
Bltouch clon:
https://www.aliexpress.com/item/3D-Touch-Auto-Leveling-Sensor-Auto-Bed-Leveling-Sensor-BLTouch-For-3D-Printers-Improve-Printing-Precision/32890485972.html?spm=a2g0s.9042311.0.0.6af163c0Pblyzz
Discarded wiring because once I provide measures and these commands always work:
BLTOUCH_DEBUG COMMAND=pin_down
BLTOUCH_DEBUG COMMAND=pin_up

foncubierta, 2020-01-27:
Starting serial connect
got {'#receive_time': 631184.990923024, u'next_clock': 333396224, u'oid': 5, u'value': 7983, '#name': u'analog_in_state', '#sent_time': 631184.979831722}
got {'#receive_time': 631185.0908369301, u'next_clock': 334996224, u'oid': 15, u'value': 7974, '#name': u'analog_in_state', '#sent_time': 631185.053696565}
got {'#receive_time': 631185.290199221, u'next_clock': 338196224, u'oid': 5, u'value': 7981, '#name': u'analog_in_state', '#sent_time': 631185.274002034}
got {'#receive_time': 631185.390550992, u'next_clock': 339796224, u'oid': 15, u'value': 7973, '#name': u'analog_in_state', '#sent_time': 631185.384404378}
Loaded MCU 'mcu' 80 commands (v0.8.0-211-gb921b8a1-20191227_013856-raspberrypi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB6,PB5,PB7 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS_twi=PC0,PC1 MCU=atmega1284p STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
Failed to verify BLTouch probe is raised; retrying.
Starting heater checks for extruder
Stats 631193.3: gcodein=70 mcu: mcu_awake=0.062 mcu_task_avg=0.000364 mcu_task_stddev=0.000482 bytes_write=3618 bytes_read=4482 bytes_retransmit=9 bytes_invalid=0 send_seq=157 receive_seq=157 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=22 stalled_bytes=0 freq=16000143 heater_bed: target=0 temp=11.6 pwm=0.000 print_time=1642.450 buffer_time=3.001 print_stall=0 extruder: target=0 temp=12.5 pwm=0.000
Stats 631194.3: gcodein=70 mcu: mcu_awake=0.062 mcu_task_avg=0.000364 mcu_task_stddev=0.000482 bytes_write=3680 bytes_read=4606 bytes_retransmit=9 bytes_invalid=0 send_seq=160 receive_seq=160 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000124 heater_bed: target=0 temp=11.4 pwm=0.000 print_time=1642.450 buffer_time=2.001 print_stall=0 extruder: target=0 temp=12.5 pwm=0.000
Stats 631195.3: gcodein=70 mcu: mcu_awake=0.062 mcu_task_avg=0.000364 mcu_task_stddev=0.000482 bytes_write=3726 bytes_read=4726 bytes_retransmit=9 bytes_invalid=0 send_seq=165 receive_seq=165 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000103 heater_bed: target=0 temp=11.8 pwm=0.000 print_time=1642.450 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.5 pwm=0.000
Stats 631196.3: gcodein=70 mcu: mcu_awake=0.062 mcu_task_avg=0.000364 mcu_task_stddev=0.000482 bytes_write=3848 bytes_read=4886 bytes_retransmit=9 bytes_invalid=0 send_seq=173 receive_seq=173 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000097 heater_bed: target=0 temp=11.5 pwm=0.000 print_time=1642.600 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.5 pwm=0.000
Stats 631197.3: gcodein=70 mcu: mcu_awake=0.062 mcu_task_avg=0.000364 mcu_task_stddev=0.000482 bytes_write=4113 bytes_read=5110 bytes_retransmit=9 bytes_invalid=0 send_seq=196 receive_seq=196 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000101 heater_bed: target=0 temp=11.5 pwm=0.000 print_time=1643.600 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.4 pwm=0.000
Stats 631198.3: gcodein=70 mcu: mcu_awake=0.016 mcu_task_avg=0.000112 mcu_task_stddev=0.000095 bytes_write=4366 bytes_read=5339 bytes_retransmit=9 bytes_invalid=0 send_seq=218 receive_seq=218 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000088 heater_bed: target=0 temp=11.6 pwm=0.000 print_time=1644.600 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.3 pwm=0.000
Stats 631199.3: gcodein=70 mcu: mcu_awake=0.016 mcu_task_avg=0.000112 mcu_task_stddev=0.000095 bytes_write=4619 bytes_read=5567 bytes_retransmit=9 bytes_invalid=0 send_seq=240 receive_seq=240 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000089 heater_bed: target=0 temp=11.5 pwm=0.000 print_time=1645.600 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.3 pwm=0.000
Stats 631200.3: gcodein=70 mcu: mcu_awake=0.016 mcu_task_avg=0.000112 mcu_task_stddev=0.000095 bytes_write=5038 bytes_read=5858 bytes_retransmit=9 bytes_invalid=0 send_seq=267 receive_seq=267 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000082 heater_bed: target=0 temp=11.5 pwm=0.000 print_time=1646.663 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.4 pwm=0.000
Stats 631201.3: gcodein=70 mcu: mcu_awake=0.016 mcu_task_avg=0.000112 mcu_task_stddev=0.000095 bytes_write=5314 bytes_read=6053 bytes_retransmit=9 bytes_invalid=0 send_seq=287 receive_seq=287 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000091 heater_bed: target=0 temp=11.5 pwm=0.000 print_time=1647.613 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.5 pwm=0.000
Stats 631202.3: gcodein=70 mcu: mcu_awake=0.016 mcu_task_avg=0.000112 mcu_task_stddev=0.000095 bytes_write=5567 bytes_read=6281 bytes_retransmit=9 bytes_invalid=0 send_seq=309 receive_seq=309 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000093 heater_bed: target=0 temp=11.7 pwm=0.000 print_time=1648.613 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.5 pwm=0.000
[... cut ...] 16 buffer_time=0.000 print_stall=0 extruder: target=0 temp=13.1 pwm=0.000
Stats 632227.3: gcodein=1307 mcu: mcu_awake=0.009 mcu_task_avg=0.000095 mcu_task_stddev=0.000080 bytes_write=29500 bytes_read=56474 bytes_retransmit=9 bytes_invalid=0 send_seq=2097 receive_seq=2097 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=706 freq=16000086 heater_bed: target=0 temp=12.1 pwm=0.000 print_time=2614.116 buffer_time=0.000 print_stall=0 extruder: target=0 temp=13.1 pwm=0.000

BlackStump, 2020-01-27:
have you read and follow the official docs?
https://www.klipper3d.org/BLTouch.html

foncubierta, 2020-01-28:
> ¿Has leído y seguido los documentos oficiales?
> https://www.klipper3d.org/BLTouch.html

Yes, and it seems that I have managed to improve something ... At least I can send the G29 (G28 + BED_MESH_CALIBRATE) the problem is that it never ends, it always ends up failing. Klipper data:
Send: G29
Recv: // probe at 50.000,30.000 is z=1.997500
Recv: // probe at 50.000,30.000 is z=1.957500
Recv: // probe at 50.000,30.000 is z=1.980000
Recv: !! BLTouch failed to deploy
Recv: !! BLTouch failed to deploy

Note, I added this to the klipper configuration:
pin_up_reports_not_triggered: False
pin_up_touch_mode_reports_triggered: False

KevinOConnor, 2020-02-02:
FWIW, the best information I have on the BL-Touch is in the documentation at: https://www.klipper3d.org/BLTouch.html .  Maybe someone else will chime in though.

-Kevin

darkporpoise, 2020-02-07:
In the attached config you seem to have an extra space character for your sensor pin config: `sensor_pin: ^ PC4`

Try: `sensor_pin: ^PC4` instead

Otherwise to me it does sounds like there is an issue with your BL-Touch clone. If it is sometimes probing correctly then it seems like you must have wired it correctly and it is passing it's own power-on self-test.

Other possible things to try:

`pin_move_time: 2.000` Start high (maybe 2 sec) and if that helps you can look to lower it.
`pin_up_reports_not_triggered:` and `pin_up_touch_mode_reports_triggered:` Try all four permutations of True/False.
`speed:` Try a faster (10?) and a slower (1?) speed.

You can also adjust the grub screw on the probe. Geetech have a [YouTube video](https://www.youtube.com/watch?v=Ix7lXvfkPEU).

KevinOConnor, 2020-03-05:
I'm going to close this ticket as it looks like the conversation has concluded.

-Kevin

closed 2020-03-05 by KevinOConnor