Klipper3d/klipper #2435: "Move out of range" after upgrade

stefanfoulis, 2020-01-27:
I upgraded klipper from `b921b8a1c1b3dcf78deba079c17d9af559f3f2a0` to `5ca132eafaaaf2b5bffc2b696d1fa4e364ffaf0b` (https://github.com/KevinOConnor/klipper/compare/b921b8a1c1b3dcf78deba079c17d9af559f3f2a0...5ca132eafaaaf2b5bffc2b696d1fa4e364ffaf0b).
I see there are a bunch of changes related to `bed_mesh`. I guessed that `min_point` / `max_point` have been renamed to `mesh_min` and `mesh_max` based on klippers log messages when it tried to connect after the upgrade.

But now whenever I start bed leveling (with bltouch) my z increases height for a bit and then stops with the error:
```
Move out of range: -20.300 17.600 5.000 [0.000]
```
and cancels the print.
I am guessing I forgot to upgrade some other configuration. Could you point me in the right direction to look further?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4117300/klippy.log)

manbehindthemadness, 2020-01-27:
I have been having this issue myself, for me it triggers after I run an autolevel without issuing a SAVE_CONFIG at the end. Once I added that to my G29 macro the problem went away. This is a workaround not a solution.

stefanfoulis, 2020-01-27:
@manbehindthemadness thanks for the workaround. I'll try that.

websheff, 2020-01-27:
same problem

manbehindthemadness, 2020-01-27:
The only other thing I can say is to be sure that the probe offset is properly accounted for as Klipper (as of my last read) isn't calculating it into the probing grid so you have to subtract that offset in the max and min grid settings by hand. Take note that it was noted that this behavior was going away for simpler configuration in the future.

websheff, 2020-01-27:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

#mesh_max:
#   Defines the maximum x,y coordinate of the mesh for rectangular beds.
#   Adheres to the same principle as mesh_min, however this will be the
#   furthest point probed from the bed's origin. This parameter must be
#   provided for rectangular beds.

Works fine:

mesh_min: 50,30
mesh_max: 185,215

stefanfoulis, 2020-01-28:
This config was failing:
```
[bltouch]
sensor_pin: ^PC15
control_pin: PC26
pin_move_time: 0.4
x_offset: 30.3
y_offset: 2.4
z_offset: 1.244

[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 10, 20
mesh_max: 180, 205
probe_count: 4, 4
```

I changed `mesh_min` and `mesh_max` generously to definitely account enough for the offset between probe and nozzle:
```
mesh_min: 50, 50
mesh_max: 170, 170
```
and now it works again.

What was confusing and why I didn't investigate in this direction, was that the carriage went up and then returned the error. So I was mainly looking for an error with `z` related settings.

manbehindthemadness, 2020-01-29:
SO I just plugged in the latest and I am having the same problem, I can manually move to the mesh_min and mesh_max locations but it's telling me the movement is out of range. NOW that being said, the documentation indicated that the Probe offset is not added to these values, did this change in the latest release?

manbehindthemadness, 2020-01-29:
Looking at the error, that seems to be exactly what's happening:
Recv: !! Move out of range: -42.200 -44.900 5.000 [0.000]
Recv: !! Move out of range: -42.200 -44.900 5.000 [0.000]

[bed_mesh]
speed: 100
horizontal_move_z: 5
mesh_min: -20,0
mesh_max: 272,142
probe_count: 5,4
algorithm: lagrange
bicubic_tension: .2
mesh_pps: 2,2
move_check_distance: 5
split_delta_z: .025
fade_start: 1
fade_end: 10

[bltouch]
x_offset: 22.2
y_offset: 44.9
z_offset: 2

jakep82, 2020-01-29:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md

jakep82, 2020-01-29:
If your mesh_min for X is -20, and your x_offset is 22.2, the nozzle would need to move to -42.2 to probe X-20.

manbehindthemadness, 2020-01-29:
yeah I got it as soon as I realized, add the offset values to the min and max values. Mystery solved.

stefanfoulis, 2020-01-29:
🙇 Thanks for all the hints and explanations 👍

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-02-14 by KevinOConnor