Klipper3d/klipper #2436: Defining a [gcode_button] causes klipper to fail with a protocol error.

manbehindthemadness, 2020-01-27:
When I define a [gcode_button] klipper fails to restart and returns a protocol error. I have experimented with the contents of the gcode_button config and it seems to have no affect. I posed the question on Discord and they pointed me here. Logs and configs attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4117512/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4117527/printer.cfg.txt)
The system is a beaglebone/Replicape B3A running the stock Debian image

Protocol error
Traceback (most recent call last):
  File "/home/debian/klipper/klippy/klippy.py", line 139, in _connect
    cb()
  File "/home/debian/klipper/klippy/mcu.py", line 521, in _connect
    self._send_config(config_params['crc'])
  File "/home/debian/klipper/klippy/mcu.py", line 454, in _send_config
    cb()
  File "/home/debian/klipper/klippy/extras/buttons.py", line 47, in build_config
    "buttons_ack oid=%c count=%c", cq=cmd_queue)
  File "/home/debian/klipper/klippy/mcu.py", line 605, in lookup_command
    cmd = self._serial.get_msgparser().lookup_command(msgformat)
  File "/home/debian/klipper/klippy/msgproto.py", line 300, in lookup_command
    raise error("Unknown command: %s" % (msgname,))
error: Unknown command: buttons_ack

FHeilmann, 2020-01-28:
`unknown command` errors usually are due to an MCU firmware that is significantly older/newer than the klippy host code that tries to interact with it. I'm not sure how this works on the beaglebone, but you can try pulling a fresh copy of the klipper repo, rebuilding and reflashing the PRU firmware. This may fix it.

manbehindthemadness, 2020-01-28:
I am on the latest software across the board. What's baffling is everything else works without issue.

KevinOConnor, 2020-01-28:
The beaglebone PRU has a very limited amount of code space (the entire program running on the PRU must not have more than 2048 instructions), so we don't compile all features into the PRU binary.  One of the features not compiled in is the buttons.c module.  If you're a programmer, you can modify src/Makefile and move the butttons.c item from the `src-$(CONFIG_HAVE_GPIO_BITBANGING)` line to the `src-y` line.  I do not know if it will fit or not, but if it does compile and flash then it should be fine.

-Kevin

manbehindthemadness, 2020-01-28:
Thanks Kevin, I will give that a spin after I backup my image.

manbehindthemadness, 2020-01-28:
Is this the output one would expect if it indeed doesn't fit?
debian@beaglebone:~/klipper$ make flash
  Compiling out/src/buttons.o
src/buttons.c:20:20: error: array type has incomplete element type 'struct gpio_in'
     struct gpio_in pins[0];
                    ^~~~
src/buttons.c: In function 'buttons_event':
src/buttons.c:35:23: warning: implicit declaration of function 'gpio_in_read' [-Wimplicit-function-declaration]
         uint8_t val = gpio_in_read(b->pins[i]);
                       ^~~~~~~~~~~~
src/buttons.c: In function 'command_buttons_add':
src/buttons.c:93:20: warning: implicit declaration of function 'gpio_in_setup' [-Wimplicit-function-declaration]
     b->pins[pos] = gpio_in_setup(args[2], args[3]);
                    ^~~~~~~~~~~~~
Makefile:66: recipe for target 'out/src/buttons.o' failed
make: *** [out/src/buttons.o] Error 1

KevinOConnor, 2020-01-29:
That looks like you tried to compile the "Linux" micro-controller code with the Makefile change (instead of just the PRU code).  As an update to my original suggestion, you could try moving buttons.c from `src-$(CONFIG_HAVE_GPIO_BITBANGING)` to `src-$(CONFIG_HAVE_GPIO)` so that both the linux and pru can be compiled.

-Kevin

manbehindthemadness, 2020-01-30:
Judging by: 
/usr/lib/gcc/pru/8.0.1/../../../pru/bin/ld: out/pru1.elf section `.text' will not fit in region `imem'
/usr/lib/gcc/pru/8.0.1/../../../pru/bin/ld: region `imem' overflowed by 308 bytes
Your original diagnosis was correct, not enough space; however, thanks for trying!

closed 2020-01-30 by manbehindthemadness