Klipper3d/klipper #2437: Sensorless homing with TMC5160 (endstops stay triggered)

BoHoehne, 2020-01-27:
For the first time sensorless homing worked,but.

On the second try only one axis (delta-printer) did home.
A query_endstops showed that two of the three endstops stay triggered.

I wasn't able to reset the triggered status.
I tried restart, restart_firmware, powercycle, 
unplugging the diag1 pin. 

Nothing helped. 

How can I solve this problem ?

[klippy_log.zip](https://github.com/KevinOConnor/klipper/files/4119097/klippy_log.zip)
[printer_cfg.zip](https://github.com/KevinOConnor/klipper/files/4119098/printer_cfg.zip)

jakep82, 2020-01-27:
I'm not sure why the endstops are staying triggered, but you should use the same run currents across all three towers.  Right now you have A and B set at 2 amps (which might be too high for your motors), and C set at 0.8 amps.  I would use the same current and SGT values for all three.

BoHoehne, 2020-01-28:
Setting the two "always triggered" steppers to 2 amps was an act of desparation.
Initially i thought the steppers weren't moving because of too low amps.
It had no influence on the endstops. 

I set all three to reasonable values (0.8/0.5). The problem persists.

BoHoehne, 2020-01-28:
How do i reset the endstop status ?

KevinOConnor, 2020-02-02:
FYI, the best information I have on sensorless homing is in the docs at: https://www.klipper3d.org/Sensorless_Homing.html

If you're not seeing the endstop status change it would generally indicate an incorrect wiring or configuration of the diag pin, and/or incorrect setting of the "sgt" setting.

-Kevin

BoHoehne, 2020-02-10:
Quadruple checked the wiring. 
Also changed to a different wire loom.

Problem persists.

KevinOConnor, 2020-03-05:
Was there any further luck with this?  I noticed a success report (possibly unrelated in #2551 ).  Also, there was a recent change to the tmc5160 current setting (PR #2534 ) - though I don't know that that pr would help here.

-Kevin

BoHoehne, 2020-03-05:
I had to take the hardware apart for spare parts. 

Another printer i build (cartesian/5160), with SPI hardwired onboard did not show this behavior with sensorless homing (10 days ago). 
Originallly this issue was on a LPC1768 with tmc5160s.

I a few days i will have the time to address the issue again on my delta project (which also uses LPC1768/9 and tmc5160s with dual mcu). 

If i don't see the problem again, i will close this issue.

KevinOConnor, 2020-04-15:
Okay, lets close this until there is further info.

-Kevin

closed 2020-04-15 by KevinOConnor