Klipper3d/klipper #2438: Bed_mesh_calibrate don't save value

Kotletowsky, 2020-01-28:
Hi,
I'm fighting with bed_mesh. First, I did bed mesh calibrate, thet save profile as "ender", then save config.
In my start g code I have profile load with given name. But during print I didn't see changes. Even when I open printer.cfg I see only some random old parameters in "default" profile, and there is no "ender" profile in it.

Arksine, 2020-01-30:
Make sure you aren't symlinking to printer.cfg, it doesn't work well with SAVE_CONFIG.

Otherwise please upload klippy.log after reproducing the issue.  If you try to load a profile that doesn't exist in printer.cfg you will get an error.

puneit-hassio, 2020-01-30:
> Hi,
> I'm fighting with bed_mesh. First, I did bed mesh calibrate, thet save profile as "ender", then save config.
> In my start g code I have profile load with given name. But during print I didn't see [...]

sometimes octoprint misbehaves. It saves the bed mesh in the back up file.  (The one with time stamp) or use another way to run Bed mesh like Octopod or create an LCD menu for bed mesh and save_config. I just know the workaround - not sure of the technical stuff behind it.

puneit-hassio, 2020-01-30:
sometimes octoprint misbehaves. It saves the bed mesh in the back up file. (The one with time stamp) or use another way to run Bed mesh like Octopod or create an LCD menu for bed mesh and save_config. I just know the workaround - not sure of the technical stuff behind it.

Hywelmartin, 2020-01-31:
Don't use octoklipper.. there's some bugs that's not fixed... There are some forks that works..

Kotletowsky, 2020-02-01:
Yes, it was problem with octoprint. Now I have dwc2, and everything is ok.

closed 2020-02-01 by Kotletowsky

Omikael69, 2021-02-03:
Hi,
I'm fighting with bed_mesh. First, I did bed mesh calibrate, thet save profile as "ender", then save config.
In my start g code I have profile load with given name. But during print I didn't see changes. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5919151/klippy.log)