Klipper3d/klipper #2440: G4 P2000 doesnt work

SphaeroX, 2020-01-28:
When I want to run G4 P2000 there is a parse error at Klipper, actually the printer should wait two seconds. Or is my code wrong?

> Error on 'N16 G4 P2000: wait in ms*96': unable to parse 2000:
> Dumping gcode input 22 blocks
> Read 3986.002963: 'M105\n'
> Read 3987.771893: 'M105\n'
> Read 3989.769061: 'M105\n'
> Read 3991.771087: 'M105\n'
> Read 3995.502060: 'N0 M110 N0*125\n'
> Read 3995.534811: 'N1 M201 X9000 Y9000 Z500 E10000*59\n'
> Read 3995.551846: 'N2 M203 X500 Y500 Z12 E120*14\n'
> Read 3995.585799: 'N3 M204 P2000 R1500 T2000*84\n'
> Read 3995.595222: 'N4 M205 X10.00 Y10.00 Z0.30 E2.50*58\n'
> Read 3995.611371: 'N5 M205 S0 T0*38\n'
> Read 3995.631385: 'N6 M107*35\n'
> Read 3995.645180: 'N7 G90*23\n'
> Read 3995.657069: 'N8 M104 S235*105\n'
> Read 3995.673719: 'N9 M140 S80*84\n'
> Read 3995.684711: 'N10 M190 S80*97\n'
> Read 4125.807192: 'N11 M109 S235*92\n'
> Read 4125.830444: 'N12 G28*32\n'
> Read 4144.698345: 'N13 G1 Y15 X15 F3000.0*96\n'
> Read 4144.717641: 'N14 G1 Z0*87\n'
> Read 4144.734511: 'N15 G1 E4 F300*24\n'
> Read 4144.844649: 'N16 G4 P2000: wait in ms*96\n'
> Read 4144.908616: 'M112\n'
> gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[15.0, 15.0, 0.0, 4.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=5.0
> Shutdown due to M112 command

jakep82, 2020-01-28:
It looks like the line says

`G4` P2000:

You need to remove the colon at the end.

SphaeroX, 2020-01-29:
> It looks like the line says
> 
> `G4 P2000:`
> 
> You need to remove the colon at the end.

Yeah sorry u are right

closed 2020-01-29 by SphaeroX