Klipper3d/klipper #2443: Calibrate the Z Probe offset already having a perfect Z height

robustini, 2020-01-29:
The PROBE_CALIBRATE command requires you to perform the test with the sheet of paper to found and save the Z offset of the probe.
It would also be useful to have a command that instead finds and saves this offset having already a Z previously detected, therefore nozzle that prints already at the right height with respect to the bed.

KevinOConnor, 2020-01-30:
If you are sure the Z calibration is correct, then you can just run PROBE followed by GET_POSITION.  The probe's z height is then the z position as reported by GET_POSITION.

-Kevin

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like the conversation has concluded.  As mentioned, it is possible to obtain the desired value using the PROBE command.

-Kevin

closed 2020-02-14 by KevinOConnor