Klipper3d/klipper #2444: SKR 1.3, RepRapDiscount Smart Controller beeping while klipper not connected

ZhenyaKa, 2020-01-29:
When connect buzzer stops to beep. If no connect - continously beep

ZhenyaKa, 2020-01-29:
[klipper.log](https://github.com/KevinOConnor/klipper/files/4130255/klipper.log)

BlackStump, 2020-01-29:
beep
beep
really try do the right thing and provide a log as per
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md
I hope you find some help, but it will not come from me ..beep beep beep to you

ZhenyaKa, 2020-01-29:
There is no any log while Klipper not connected to board!!!

KevinOConnor, 2020-01-29:
You must attach the Klipper log file from the event in order to use this github issue tracker - as described at: https://www.klipper3d.org/Contact.html

-Kevin

closed 2020-01-29 by KevinOConnor

ZhenyaKa, 2020-01-29:
What a log??? Are you stupid like a your bot???
Board beeping BEFORE CONNECTING KLIPPER TO BOARD!!!

BlackStump, 2020-01-29:
You are arrogantly saying it has nothing to do with klipper but persistent in asking your question here so really who is stupid?

ZhenyaKa, 2020-01-30:
Okay, "smart" guys!

We have a board. We have a printer. We press power button to switch it on (board on printer+orange pi). We got peeeeeeeeeeeeeeeeeeeeeeeeep while Armbian booting. All the fucking time printer peeeeeeps.
We have no any log at that time, because beeeeeeep

BlackStump, 2020-01-30:
Make menuconfig
 select "Enable extra
 low-level configuration options" and configure "GPIO pins to set at
 micro-controller startup" to "!P1.30"
I have no idea how you have configured your board because you did not provide a log!!!!!!!!!! maybe that is your beeper pin maybe it isn't but you are a smart guy figure it out.

ZhenyaKa, 2020-01-30:
Oh GOD! Thank you a lot!!! Beeeeping gone away!