Klipper3d/klipper #2445: Setting extruder calibration from gcode or terminal

dewi-ny-je, 2020-01-29:
Different filaments require different pressure advance values, different steps/mm calibration and potentially different filament diameters (2.85 mm vs 3.00 mm, for example).

Pressure advance can be set via the slicer using SET_PRESSURE_ADVANCE ADVANCE=xxx, but I couldn't find any way to change the extruder calibration and the filament diameter from the terminal.

This is important to define filament profiles in the slicer.

I can change the extrusion factor to compensate manually, but it looks like an hack. Is there any other way to do it?

KevinOConnor, 2020-01-30:
Currently, the only way to change the extrusion rate is via the M221 command (set extrude factor override percentage).  The filament diameter in the config is only used for sanity checks, so you can safely set that to the largest meaningful value for your extruder.

-Kevin

dewi-ny-je, 2020-01-30:
However changing the extrusion multiplier affects the accuracy of the filament length indicated in the slicing summary.

So let me change the ticket from support request to feature request:

It is common to use multiple filaments, which have likely similar diameter, but often very different hardness and therefore require different steps/mm calibration (the hobbed bolt grips them at different depths). Hacks such as changing the extrusion multiplier or the filament diameter in the slicer software are possible, but they appear to be just that, workarounds.

Since one of the advantages of Klipper is not having to reflash firmware, therefore making the change of settings much easier, I think it would be beneficial for the users and AA normal evolution for the software to offer some extruder parameters (all those which are filament-dependent) turnable from the command line, so that the users can take care of everything in the slicer.
Nozzle diameter, while not used in the calculations, could be made tunable or not, since it is indeed a much less frequently changed part anyway.

KevinOConnor, 2020-02-02:
Understood.  A developer will likely need to step forward to implement and submit it.

FYI, there may be some related discussion at #950 .

-Kevin

KevinOConnor, 2020-03-05:
I'm going to close this as it looks like the conversation has concluded.  If a developer wishes to implement the above then let me know.

-Kevin

closed 2020-03-05 by KevinOConnor

ld3300, 2020-03-07:
I have submitted a pull request for tunable extruder step distance. #2598 
It might look at the nozzle diameter if it seems tenable.

ld3300, 2020-03-07:
@KevinOConnor 
sorry, forgot to reach out and let you know I might tackle this.  Honestly I wasn't sure if I could so I decided to start and figured it out.  There was some talk of breaking the step distance down into a set of values.  Let me know if this was the right track or if there is a branch I should do this from instead.
Thanks,
-David

ld3300, 2020-03-26:
Looking at it, I don't think there is much value in making nozzle and filament size adjustable.  It could be done, but the only thing they are used for is calculating the extrude_cross_section.  So, I'm not going to touch those parts unless for some reason a significant demand pops up.