Klipper3d/klipper #2446: how not disable motors?

jornada812, 2020-01-30:
I have a delta printer with 0.9 motors and they spin easily. If I lift the extruder into the home after turning on the printer, then after a while the motors turn off and the extruder falls from a height of 0.5 meters. It hurts. How to avoid this? I do not want to automatically turn off the motors

[octoprint.log](https://github.com/KevinOConnor/klipper/files/4131689/octoprint.log)

jakep82, 2020-01-30:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L437

jornada812, 2020-01-30:
[idle_timeout]
gcode:
#   A list of G-Code commands to execute on an idle timeout. See
#   docs/Command_Templates.md for G-Code format. The default is to run
#   "TURN_OFF_HEATERS" and "M84".
timeout: 10

it does not work. Motors shut down anyway

Hywelmartin, 2020-01-30:
stepstickprotectors.. they make my effector harder to drag around...

jornada812, 2020-01-30:
These are crutches. In marlin, rrf these things are solved easily and simply. Is there really such a thing here? In general, why does the firmware decide to turn off the motors for me? This is the wrong behavior.

jakep82, 2020-01-30:
I provided you a solution that should work.  You can set a custom idle timeout.  Also the log you provided is the Octoprint log.  If you're still having an issue, please supply a Klipper log.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#issue-reporting

closed 2020-02-06 by klipper-gitissuebot