Klipper3d/klipper #2447: Not able to install on Ender 3X

titogarrido, 2020-01-30:
Hi folks!

My device is an Ender 3X, never flashed a bootloader on it. It is pretty much stock and I am trying to install Klipper for the first time.

My printer shows version 1.1.6.2 Ender-3 and when I try to flash the firmware I am getting:

```
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 via avrdude
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x46
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x61

avrdude done.  Thank you.

make: *** [src/avr/Makefile:34: flash] Error 1
pi@octopi:~/klipper $ cat /tmp/klippy.log
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-283-ge3dde199'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.16 (default, Apr  6 2019, 01:42:57) \n[GCC 8.2.0]'
Building C code module c_helper.so
Start printer at Wed Jan 29 20:59:28 2020 (1580342368.6 44290.4)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 98, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 120, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 215, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 104, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg
```

puneit-hassio, 2020-01-30:
did You flash a bootloader first?

titogarrido, 2020-01-30:
> did You flash a bootloader first?
> […](#)
> Sent from my iPhone

Good point, I have the Ender 3 for a week only... It is pretty much stock, but I guess you are right, I didn't flash a bootloader yet. I will try to find a tutorial that uses a raspberry pi to flash.

Thank you!

KevinOConnor, 2020-01-30:
I'm not sure what type of micro-controller is in the ender 3x, but if it is an atmega1284p micro-controller then try the avrdude command mentioned at the top of: https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg

-Kevin

titogarrido, 2020-01-31:
Hi Kenvin, I have tried it failed... seems that I have to setup some wiring before flashing the bootloader. I will give a try on Saturday.

Thanks!

closed 2020-02-06 by klipper-gitissuebot