Klipper3d/klipper #2448: SKR 1.3 + TMC 2209 UART "Unable to read tmc uart 'stepper_c' register IFCNT"

matschi140, 2020-01-30:
Hello,

I converted my Flsun QQ to Bigtreetech SKR 1.3 + TMC2209 Stepper also from Big tree tech so there should no soldering be necessary if  i am right.

For testing i installed only the C Stepper and configured only this one in the cfg according to the generic-bigtreetech-skr-v1.3.cfg + example-delta.cfg but keep getting this error message.

With an A4988+ corresponding jumper set the motor moves as it should.

Can anybody help me with my issue? I will attach the log later when i am at the printer

Thanks

matschi140, 2020-01-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4135407/klippy.log)

here is the log, does it play any role where in the config file i put in  the TMC2209 section?

Hywelmartin, 2020-01-30:
[tmc2209 stepper_b] <------ where did you but the stepstick??? Stepper c???
uart_pin = P1.10
microsteps = 16
interpolate = True
run_current = 1
hold_current = 0.650
sense_resistor = 0.110
stealthchop_threshold = 30

matschi140, 2020-02-02:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4144372/klippy.log)

I installed all 4 BTT TMC 2209 now, i also tried with Marlin 2.0 and get same error on stepper C in Marlin called stepper Z.

i tried swapping Stepper Drivers around but the fault remained on the stepper C

So in my opinion it must be something related with the SKR 1.3 Board...

What do you think, please check the log.

thank you for your time!!

matschi140, 2020-02-02:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4144830/klippy.log)
So i Think the fault is really the SKR C stepper driver socket. I moved the driver and the C Motor to the E1 Socket and now i have no errors. 

But the next Problem now if i start a Print, it heats up all heaters to the target temp but after that nothing happens?!?! 

What is wrong? pls check my log.

Thanks again

vladT0, 2020-02-03:
I had very similar issue with TMC2208. See https://github.com/KevinOConnor/klipper/issues/542

matschi140, 2020-02-03:
Thanks VladT0!

But i already got a connection to all drivers and all is working now from Terminal homing, moving axis, heating,... but when i start a print it stops after the heating, extruder and bed stays at printing temp but last command i see in Terminal is M119 T220 after that nothing happens.

WHAT CAN CAUSE THIS PROBLEM??

ld3300, 2020-02-04:
Will you try replacing M109 S220 T0 in the gcode file with M109 S220 and let me know if that makes a difference? Do the same with the M104 statement right before it

matschi140, 2020-02-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4154191/klippy.log)

with A4899 it started printing, but with the TMC2209 in UART mode not, can you see anything else wrong in my log? today i ordered a new skr 1.3 board as i think something got damaged when i had put one TMC2130 the wrong way in in stepper C Port....

vladT0, 2020-02-04:
According latest attached log [tmc2209 stepper_a] [tmc2209 stepper_b] [tmc2209 stepper_c] is not configured. Or did you attach log for A4899 ?

matschi140, 2020-02-05:
I installed a new SKR 1.3 today and it is Working now, but i had to to an PID Calibrate it only worked after that. Maybe the old Board will work also now, I noticed that it started to print only after the extruder temp got within 0.2C of target temp....

Thanks for your support!

closed 2020-02-05 by matschi140

welo78, 2020-12-30:
@matschi140 how did you solve the stepper_c issue ? i am having the same one on my stepper_z socket with skr v1.3 ... everything else works, others stepper sockets or steppers dont have any issues

matschi140, 2020-12-30:
@welo78 for me it was the driver socket on the board. as i said, i switched the socket to E1 then all worked, and still working today.

welo78, 2020-12-30:
> @welo78 for me it was the driver socket on the board. as i said, i switched the socket to E1 then all worked, and still working today.

the problem is that I have dual Z configuration, so idk how should I tackle this ... either splize the Z1 stepper driver to control both motors or idk what to do