Klipper3d/klipper #2451: MCU Timeout on wait for 'identify_response' response - No connection at all

miltonquranda, 2020-01-31:
I have been trying to connect klipper to my SKR 1.3 board from my Raspberry PI 3B+ and also from orange pi lite 2. Klipper does not connect. 

OctoPrint connects with the same port to SKR board without any issues. I tried every thing every available option, even UART connection, Octoprint connects to the board with both USB and UART but klipper does not. I feel there is a configuration issue! but not sure what it is.. Please help I am out of options.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4137480/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4137482/printer.cfg.txt)

BlackStump, 2020-01-31:
https://www.klipper3d.org/Config_Changes.html
20191126

miltonquranda, 2020-01-31:
My mistake.. firmware was miss-spelled when i transferred it to sdcard

closed 2020-01-31 by miltonquranda