Klipper3d/klipper #2452: error: Can not update MCU 'mcu' config as it is shutdown

miltonquranda, 2020-01-31:
I read through some of the responses related to the issue but it did not help.

the error pops up when serial connection happens

Loaded MCU 'mcu' 77 commands (v0.8.0-283-ge3dde199-20200131_051303-octopi / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 RECEIVE_WINDOW=192 SERIAL_BAUD=250000 BUS_PINS_ssp1=P0.8,P0.9,P0.7 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_serial=P0.3,P0.2 MCU=lpc1768 STATS_SUMSQ_BASE=256 CLOCK_FREQ=100000000
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 139, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 505, in _connect
    config_params = self._send_get_config()
  File "/home/pi/klipper/klippy/mcu.py", line 493, in _send_get_config
    self._name,))
error: Can not update MCU 'mcu' config as it is shutdown

I tried FIRMWARE_RESTART maytimes but the error persists.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4137951/printer.cfg.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4137956/klippy.log)

KevinOConnor, 2020-01-31:
That error is typically cleared by a FIRMWARE_RESTART command.  If that command does not clear the error, then you'll need to attach a log file of a session that includes that FIRMWARE_RESTART command.

-Kevin

miltonquranda, 2020-01-31:
@KevinOConnor  updated log with FIRMWARE_RESTART command executed few times
[klippy.log](https://github.com/KevinOConnor/klipper/files/4140172/klippy.log)

Octoprint log:
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6a2f3cd0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Send: N2 M21*18
Recv: ok FIRMWARE_VERSION:v0.8.0-283-ge3dde199 FIRMWARE_NAME:Klipper
Recv: ok
Send: M105
Recv: ok T:0
Send: FIRMWARE_RESTART
Recv: ok
Send: M105
Recv: // Klipper state: Disconnect
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: G91
Recv: // Can not update MCU 'mcu' config as it is shutdown
Send: G28 X0 Y0
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Can not update MCU 'mcu' config as it is shutdown
Recv: ok
Send: G90
Recv: // Can not update MCU 'mcu' config as it is shutdown
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Can not update MCU 'mcu' config as it is shutdown
Recv: ok
Recv: // Can not update MCU 'mcu' config as it is shutdown
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: !! Can not update MCU 'mcu' config as it is shutdown
Recv: ok
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: FIRMWARE_RESTART
Recv: ok
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: FIRMWARE_RESTART
Recv: ok
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0

KevinOConnor, 2020-01-31:
Oh, you're using serial on an lpc176x chip.  You should add "restart_method: command" to the mcu section of your config (see config/example.cfg for the details).

-Kevin

miltonquranda, 2020-01-31:
Thanks @KevinOConnor that worked but now I am getting ADC out of range. Not sure which one its complaining about.. Is it the heated bed? 

Stats 1987.9: gcodein=5 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=2079 bytes_read=4476 bytes_retransmit=9 bytes_invalid=0 send_seq=179 receive_seq=179 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100000082 heater_bed: target=0 temp=0.0 pwm=0.000 print_time=0.000 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=0.0 pwm=0.000
Stats 1988.9: gcodein=5 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=2085 bytes_read=4642 bytes_retransmit=9 bytes_invalid=0 send_seq=180 receive_seq=180 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001257 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=0.000 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000 extruder1: target=0 temp=-67.5 pwm=0.000
MCU 'mcu' shutdown: ADC out of range

[klippy.log](https://github.com/KevinOConnor/klipper/files/4140379/klippy.log)

miltonquranda, 2020-01-31:
seems extruder1 temperature is showing as -72.5 

extruder1: target=0 temp=-72.5 pwm=0.000

Now i have 2 in and 1 out.. So there is no extruder 0 and extruder 1 heater. Just one heater connected for both. How do i handle this in the config?

jakep82, 2020-01-31:
You need to define a shared heater.  See the link.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L681

miltonquranda, 2020-01-31:
That worked. Issue has been resolved. Closing this ticket.

closed 2020-01-31 by miltonquranda

FrincuFPV, 2021-01-13:
I try on ender 3 v2 v4.2.2 with pi4 2gb and no results. 1w after still nothing. Any advice?

jakep82, 2021-01-13:
@FrincuFPV Replying to an unrelated issue that was closed almost a year ago is not the appropriate way to request help.

FrincuFPV, 2021-01-14:
Well i don't understand how to setup this part here. I try all the combination and nothing. Last night i managed to get some Bin file... But after putting it on the printer start the buzzer and it does not stop 🤔. I will try again later today. And i don't see any videos or how to v4.2.2 board. Thanks again. 
[image]