Klipper3d/klipper #2455: [Pitch] Bed leveling features from Marlin's UBL

GarthSnyder, 2020-02-01:
I'm in the process of converting from Marlin, and the only things I miss are some of Marlin's bed leveling features. In particular, there are some really nice aspects of Marlin's Unified Bed Leveling. If I were to work on implementing similar features in Klipper, would you be receptive to a PR?

The main feature is that UBL can save and restore a high-resolution bed map that describes the intrinsic geometry of the build plate (cupped, rippled, etc.). It also has a an operation (G29 J) that does a quick probe of the build plate and then tilts the current mesh to be consistent with what it finds. The idea is that you create the detailed map once, then just do a quick three- or four-point probe before each print job to adjust for the current attitude of the build plate. Much faster, just as accurate, and you're free to improve the physical leveling of the bed.

The other thing that Klipper doesn't appear to have (though I may have missed it) is an incremental approach to building meshes. Logically, UBL meshes cover the entire build plate and the areas that are unreachable by the probe are just marked as "no data", which is a first-class mesh entry. You can fill in holes by hand, just leave them be, or do an incremental probe operation to pick up additional data.

For example, I have a dual BLTouch mount that allows the probe to be positioned in two different quadrants relative to the nozzle. You can build an initial mesh in one position, then move the probe to the second position and fill in any holes that have now become accessible. (Even together, the two positions don't cover 100% of the bed, but they come pretty close.)

More generally, UBL uses a data-centric model. Where Klipper rolls interpolation into the probing operation, UBL keeps it separate. There's a "fill in this mesh as best you can" operation that's separate from probing and storage. You can keep your meshes in the form of raw data and just use interpolation to flesh them out at the point of use, without disturbing any of the underlying information.

KevinOConnor, 2020-02-03:
Hi.

FYI, combining bed_mesh with tilt was discussed a bit in PR #2431 .

To your high-level question, "would you be receptive to a PR?", I can say that I do regularly review and merge new features submitted as PRs.  The current bed_mesh code is written and maintained by @Arksine , so he's probably the best person to coordinate with for any mesh discussions and enhancements.

-Kevin

GarthSnyder, 2020-02-03:
Ah, wonderful! It looks like at least that part is well in progress. (Was the intended xref to PRs #2018 and #2107?) I will switch this discussion over to #2107.

closed 2020-02-03 by GarthSnyder

KevinOConnor, 2020-02-03:
Yes, #2107 - sometimes github does some weird autocomplete on my PR references..

-Kevin