Klipper3d/klipper #2456: BL-Touch tip, moves several times while probing, up and down in the Z-movement.

Metti93, 2020-02-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4142503/klippy.log)

Here a little Video of the issue:  https://imgur.com/a/pT7qvFu

While my Ender 3 is measuring the bed using the "G29" command, the tip of the BL-Touch panicked up and down during the Z movement. Which leads to my bed being measured sluggishly.
How so?

Metti93, 2020-02-01:
I have now adjusted the "pin_move_time", under [bltouch], to 1.0. 
As well as "horizontal_move_z", in [bed_mesh], adjusted to 10.

Now he seems to have fewer problems.

Metti93, 2020-02-01:
Nope. A measurement of 20 fails on every case and measures a wrong value

KevinOConnor, 2020-02-14:
FWIW, I personally don't have any insight into the cause of your issues.  Perhaps someone else will chime in.

-Kevin

BlackStump, 2020-02-14:
Creality board + bltouch = problems 

https://github.com/KevinOConnor/klipper/issues/2321

DrShats, 2020-03-06:
Any news on the case? Experiencing same problem on mega2560+rpi3b and on mega2560 + rpi3b+

KevinOConnor, 2020-03-21:
I'm going to close this as it looks like the conversation has concluded.  FYI, there are several BL-Touch fixes in the most recent code.  Also be sure to follow the directions in docs/BL-Touch.md.

-Kevin

closed 2020-03-21 by KevinOConnor