Klipper3d/klipper #2460: Incorrect mesh position calculation

jbbandos, 2020-02-02:
I'm having issues getting the mesh calibration to work. I have an ender 3 that I used before with klipper, and now that I've upgraded to an SKR Mini E3 board and a BLTouch I wanted to use klipper instead of merlin again.
It seems to be working except for the bltouch and mesh. For the mesh, I see it apparently adding the mesh_min y to the x coordinate - eg, instead of starting at 30,30 it starts at 60,30. If I change x to 78, it starts at 108, etc. That means that not only are all probe points shifted, but the last calculated probe point, at least, is out of bounds.
I tried setting y to 0, but then I get a "Failed to home probe: Timeout during endstop homing" error. I can see the head moving, homing, and it is only after when it moves to what should be 30,0 that it fails to detect the bed and errors out.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4144396/klippy.log)

ld3300, 2020-02-02:
The mesh calculation changed.  It is now based on the position of the probe.  You need to calculate the probe position based on the x and y offset of your BLtouch.  See the config changes and example extras.

[Config_Changes](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md)

[example-extras](https://github.com/KevinOConnor/klipper/blob/c36fe76d692754d0244059a729e147c688c96062/config/example-extras.cfg#L60)

jbbandos, 2020-02-03:
Ok, I could understand that, it explains why 0,0 as the mesh_min would start probing outside the bed. I'd just add/subtract the probe offsets, and it is ok. 
However, what I am seeing is a different issue - the Y mesh_min being added to the X mesh_min, and the mesh probing starting point being (X+Y, Y)

ld3300, 2020-02-03:
If you redo the mesh and save_config does it still do that?

BlackStump, 2020-02-03:
If using "probe:z_virtual_endstop" then do not define a position_endstop in
 the stepper_z config section.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L541

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L606

Arksine, 2020-02-03:
Your bltouch x_offset is being subtracted from the "X" mesh_min to produce coordinates that will move the probe to the mesh_min coordinate.
```
[bltouch]
sensor_pin = ^PC2
control_pin = PA1
pin_up_reports_not_triggered = False
pin_up_touch_mode_reports_triggered = False
x_offset = -30.13
y_offset = -0.5
z_offset = -2.56
pin_move_time = 1.5
speed = 5.0
samples = 2
sample_retract_dist = 8.0
```
```
Index |  Tool Adjusted  |   Probe
  0   | (80.1, 30.5)    | (50.0, 30.0)
  1   | (125.1, 30.5)   | (95.0, 30.0)
  2   | (170.1, 30.5)   | (140.0, 30.0)
  3   | (215.1, 30.5)   | (185.0, 30.0)
  4   | (260.1, 30.5)   | (230.0, 30.0)
  5   | (260.1, 80.5)   | (230.0, 80.0)
  6   | (215.1, 80.5)   | (185.0, 80.0)
  7   | (170.1, 80.5)   | (140.0, 80.0)
  8   | (125.1, 80.5)   | (95.0, 80.0)
  9   | (80.1, 80.5)    | (50.0, 80.0)
  10  | (80.1, 130.5)   | (50.0, 130.0)
  11  | (125.1, 130.5)  | (95.0, 130.0)
  12  | (170.1, 130.5)  | (140.0, 130.0)
  13  | (215.1, 130.5)  | (185.0, 130.0)
  14  | (260.1, 130.5)  | (230.0, 130.0)
  15  | (260.1, 180.5)  | (230.0, 180.0)
  16  | (215.1, 180.5)  | (185.0, 180.0)
  17  | (170.1, 180.5)  | (140.0, 180.0)
  18  | (125.1, 180.5)  | (95.0, 180.0)
  19  | (80.1, 180.5)   | (50.0, 180.0)
  20  | (80.1, 230.5)   | (50.0, 230.0)
  21  | (125.1, 230.5)  | (95.0, 230.0)
  22  | (170.1, 230.5)  | (140.0, 230.0)
  23  | (215.1, 230.5)  | (185.0, 230.0)
  24  | (260.1, 230.5)  | (230.0, 230.0)
```
You have an x_offset of -30.1.  So as you can see from the log's generated points, when the Probe is at X=50 (mesh_min), the tool will move to X=80.1, and so on.  It only looks like its X+Y because your Y mesh_min is 30.

jbbandos, 2020-02-03:
Ok, I finally understood it. Seems confusing have to go back to the reality that the probe is not at 0,0 when doing the mesh calculations, instead of abstracting and basing the points on the nozzle position, but I understand. It should be clearer in the documentation, maybe pointing to @Arksine great explanation above?

@ld3300 I couldn't finish the mesh, as it would end out of the plate and abort on the first line

closed 2020-02-03 by jbbandos

Arksine, 2020-02-03:
FWIW, I did my level best at creating full documentation for the bed_mesh module, including visualizations.  

https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md

Ultimately mesh_min and mesh_max should represent the coordinates of the actual mesh location (vs the tool location as it was before).  It seems like regardless which way we go with this there will be some confusion, however the current method is consistent with other firmware so hopefully going forward it will be the best.

jbbandos, 2020-02-03:
@Arksine You did a great work, but for old and slow brains like mine you could probably make it more evident that the used value is now the "tool adjusted" one, adding the offsets to the mesh_min/mesh_max, and that the user should make sure that the adjusted mesh_max/min are inside the bed. 
Anyway, I've closed this issue as it is clearly PEBKAC.