Klipper3d/klipper #2461: Octoprint Reboot Required?

radensb, 2020-02-02:
This is a minor annoyance, but potentially indicates that something is not right. Every time I power cycle the printer, I have to reboot the Pi running Octoprint. If I do not, Octoprint will show connected until I try to send a command or get the status from the firmware. It will then give me the "Cannot communicate with MCU" error. I have tried restarting Octoprint, resetting the firmware and Host in the Octoklipper plugin, and logging into a Pi terminal and stop/start/restarting Klipper manually. Nothing works until I issue a reboot to the Pi, then the printer comes up and works as expected.

I am currently experimenting with my config file in a way that requires me to power off the machine if it doesn't behave properly, and that forces a Pi reboot every time. Any clue to why this is needed and have others experienced this? Is this normal behavior or am I not restarting the right thing? BTW, I am running v0.7 from around June. Looking to update once I work out some of my current settings, but I didn't see anything related to this issue in the change log for v0.8.

Thanks!
[klippy.log](https://github.com/KevinOConnor/klipper/files/4145433/klippy.log)

ld3300, 2020-02-03:
I have my MCU connected to my pi with a USB cable that has the power line cut.  After a print and a certain timeout my octoprint instance kills a relay that turns off the power to the entire machine except the pi itself.  When I trigger the relay back on I have no problems connecting.

JohnEdwa, 2020-02-03:
It's possible that the printer gets a new USB ID when it reconnects, [you should follow this](https://www.klipper3d.org/FAQ.html#wheres-my-serial-port) and change it to point to the actual device instead.

Also, if you want Klipper to auto-connect [you have to do this workaround.](https://github.com/KevinOConnor/klipper/issues/835#issuecomment-441186301) Otherwise, Klipper will connect to the printer, but because it was shut down unexpectedly it'll be in an error state that requires you to issue the `RESTART` command first.

Arksine, 2020-02-03:
This sounds like it is more of a linux issue than a Klipper issue.  When you are powered up and connected, ssh to the Pi and enter the dmesg command.  You should see something like the following: 
```
[79639.509084] usb 1-1.4: new full-speed USB device number 27 using dwc_otg
[79639.643663] usb 1-1.4: New USB device found, idVendor=2341, idProduct=abcd, bcdDevice= 1.00
[79639.643673] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[79639.643678] usb 1-1.4: Product: samd21g18a
[79639.643682] usb 1-1.4: Manufacturer: Klipper
[79639.643687] usb 1-1.4: SerialNumber: 1283596E4E375051202020502E1D0CFF
[79639.644464] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```
Note that if you are using a mega/avr based controller then the Manufacturer not be Klipper.   Also take note of the last timestamp, because you will want to compare it later.  Power cycle the device, the enter dmesg again.  You should see a message like this:
```
[79638.450109] usb 1-1.4: USB disconnect, device number 26
```
Followed by a reconnection with newer timestamps.  If it does not, then your board isn't re-enumerating properly after a power cycle.

radensb, 2020-02-03:
@JohnEdwa This worked perfectly! Thanks!

closed 2020-02-03 by radensb