Klipper3d/klipper #2462: Pressure Advance Calibration formula incorrect?

t3mulligan, 2020-02-03:
On this page, https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md, there is a formula to calculate your pressure advance setting by measuring the calibration tower. 
That formula:
`The pressure_advance value can then be calculated as pressure_advance = <start> + <measured_height> * <factor>. (For example, 0 + 12.90 * .020 would be .258.)`
I believe, is incorrect. I'm pretty sure this would work, if the pressure advance changed every millimeter, but it actually changes every layer. I think the correct formula is:
`The pressure_advance value can then be calculated as pressure_advance = <start> + <measured_height>/<layer_height> * <factor>. (For example, 0 + 35/0.25 * .004 would be 0.56.)`

drphil3d, 2020-02-03:
Oh that's interesting, it would explain some things

t3mulligan, 2020-02-03:
@klipper-gitissuebot My dude, there are no relevant logs, its an issue with the documentation.

jakep82, 2020-02-03:
I believe the the pressure advance factor is per millimeter, and it adjusts based on layer height. I recall watching during a calibration print and following the math. I don't believe the formula is wrong.

t3mulligan, 2020-02-03:
@jakep82 It says this right below the formula.

> The above TUNING_TOWER command instructs Klipper to alter the pressure_advance setting on each layer of the print.

So the issue is instead, the quote above, the line after the formula, if it is in fact per millimeter. I'm going to test both tonight to be certain.

Bao-gxg, 2020-02-03:
It looks like the documentation is consistent. If I'm reading the code right, it _does_ ["alter the pressure_advance setting on each layer"](https://github.com/KevinOConnor/klipper/blob/9aab58605b22a3c6009890a3ded2f1ed7af5b6f7/klippy/extras/tuning_tower.py#L63) but that setting is calculated based on [the current z-height](https://github.com/KevinOConnor/klipper/blob/9aab58605b22a3c6009890a3ded2f1ed7af5b6f7/klippy/extras/tuning_tower.py#L50), rather than layer.

I haven't read the source-code though, so I might be making some assumptions.

t3mulligan, 2020-02-03:
@Bao-gxg Gotcha. Perfect, thanks!

closed 2020-02-03 by t3mulligan