Klipper3d/klipper #2467: Endstop phase calibration does not handle multistepper axes correctly

GarthSnyder, 2020-02-04:
My printer has dual, independent Z motors, so I have both a `[stepper_z]` section and a `[stepper_z1]` section, with a corresponding `[tmc2208]` section for each. There is a separate endstop for each Z motor, although I'm not sure whether that matters for purposes of this report.

I wanted to do endstop phase calibration on the Z motors, but ENDSTOP_PHASE_CALIBRATE only reports a value for stepper_z, even though both `stepper_z` and `stepper_z1` have their own `[endstop_phase]` sections.

Just glancing over the code, it appears that phase tracking is done per-rail rather than per-motor; I'm assuming rail = axis in this context. Isn't that incorrect? It seems that this would leave every multi-motor axis with one phase-calibrated motor and N that are allowed to just dangle randomly.

A subsidiary issue is that you can't even run `ENDSTOP_PHASE_CALIBRATE STEPPER=stepper_z`, because the phase tracker no longer seems to recognize the motor after it has been rolled up into its rail. A plain `ENDSTOP_PHASE_CALIBRATE` does report a value for stepper_z, however.

[printer.cfg](https://gist.github.com/GarthSnyder/ee0483da2921ff8b9d38644a1bf8b97a)
[klippy.log](https://gist.github.com/GarthSnyder/5d9186280922c2b5fba3c5ef3ba7a675)

KevinOConnor, 2020-02-05:
The endstop phase code wasn't really designed for multiple steppers on the same axis.  The code works by adjusting the logical position of the stepper after the stepper completes its homing move.  That scheme doesn't work when multiple steppers and multiple endstops are on a single axis (instead, the code would need to physically move the steppers - much like how z_tilt and quad_gantry_level work).

FYI, many users with multiple z steppers end up doing stepper adjustments via z_tilt.

Separately, in the future, it's easier for us if you attach the log file here (not link to it).

-Kevin

GarthSnyder, 2020-02-05:
Thanks, I will attach directly in the future.

I have a fairly vanilla I3-style setup, only with each side of the X axis on a separate Z motor. (It's an Artillery Sidewinder X1.) The docs for z_tilt seem to be phrased in terms of a motorized bed that raises and lowers on multiple Z steppers, but I would imagine it's adaptable to this situation.

I'm trying to avoid the approach commonly used in Marlin (the G34 command) where the X axis is tilted to match the current plane of the bed, whatever that happens to be. It seems easier and more trouble-free to make X actually level and leave the bed as the only element that tilts.

One issue this brings up is that there doesn't seem to be a standard way in Klipper to specify a fixed offset between the two Z endstops. I can measure the skew between them pretty accurately, but it's hard to actually move the second endstop with submillimeter precision to make it exactly match the first endstop. As with phase calibration, correcting for this seems like this would have to involve moving the motors.

KevinOConnor, 2020-02-06:
Right, adding a logical endstop position for each endstop would then also require a move of the Z steppers (like z_tilt does).  It's certainly possible (and useful), but it would need to be implemented.

-Kevin

GarthSnyder, 2020-02-06:
Thank you. I may take a stab at this, but I'll close this issue for now.

closed 2020-02-06 by GarthSnyder

nicedevil007, 2020-03-06:
@GarthSnyder do you have a working config for the Artillery Sidewinder X1? I'm googling around since 2-3 days now to find any config for klipper. Hope you read this and can help me :)

GarthSnyder, 2020-03-07:
@nicedevil007, I'm using an SKR V1.3 mainboard, but you're welcome to take a look. I also have dual Z endstops and a BLTouch. The screws_tilt_adjust section is in probe coordinates, which is not the case for mainline Klipper - you'll have to convert to nozzle coordinates.

I'm continuing to tweak things, but the system has been very solid with this configuration so far.

There's an "Artillery Sidewinder X1 mod/fix international group" on Facebook that has a printer.cfg for the stock MKS Gen L in the files section. 

[printerSKRV1.3.cfg.zip](https://github.com/KevinOConnor/klipper/files/4301367/printerSKRV1.3.cfg.zip)

GarthSnyder, 2020-03-07:
Oh, also I have 0.9 degree steppers on X and Y, with TMC2208 drivers for all motors.

nicedevil007, 2020-03-07:
@GarthSnyder Thank you so far. Since I have everything stock besides the metall idler and the print bed I would go with the file from the facebook group. (I have to find the file first haha, I wasn't using facebook since 4 years now o_O)

rbn920, 2020-05-08:
> @GarthSnyder Thank you so far. Since I have everything stock besides the metall idler and the print bed I would go with the file from the facebook group. (I have to find the file first haha, I wasn't using facebook since 4 years now o_O)
Any luck tracking it down? I don't have much interest in re-joining facebook.

rbn920, 2020-05-08:
@nicedevil007 any luck tracking it down? I have no interest in re-joining facebook.