Klipper3d/klipper #2468: CR10s blank LCD and mesh_min error

jdbrassell, 2020-02-04:
Hello all,

This is my first time attempting to set up Klipper.  The process was going ok until I started Klipper after using a canned firmware (CR10s).  It connected just fine but instead of reporting a temp it throws a T:0 which i guess means there's an issue.  I put in a G28 to see what's going on and get:

Recv: // No option 'mesh_min' in section: 'bed_mesh'
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! No option 'mesh_min' in section: 'bed_mesh'

I ALSO have a blank LCD on my printer that wasn't there until I started Klipper with the CR10s profile.

For what it's worth i'm using a TH3d 2 into 1 board and I have a second extruder but no idea how to configure it.  But that's why it's commented out in my file. 

I've been trying to skim the interwebs for ideas on how to resolve these issues but am coming up blank :(  

Thank you for your assistance in advance

[klippy.log](https://github.com/KevinOConnor/klipper/files/4151088/klippy.log)
 -->

jdbrassell, 2020-02-04:
i'm actually on my way to solving this I think.  I found that I had played with the printer.cfg file and added values to it before going into the CR10s file where I had been working.  When I started Klipper I guess I assumed it would use the file I had been playing with but I guess not and I have no idea how to tell it to use that file for the config.

I copied my CR10s config into the printer.cfg file and flashed that and am making progress now

ld3300, 2020-02-04:
Hi @jdbrassell 
It looks like your build is dirty.  Can you try to remove and redo the klipper install?

Did you have any issues flashing the klipper code to the control board?

David

jdbrassell, 2020-02-04:
> Hi @jdbrassell
> It looks like your build is dirty. Can you try to remove and redo the klipper install?
> 
> Did you have any issues flashing the klipper code to the control board?
> 
> David

@ld3300 First off, thank you for the reply,

i'll start digging into how to do that.  I reflashed Marlin back to my printer to make sure i didn't brick it.  I'm a noob to this so i have no idea how to remove it from the pi itself but my google fu is strong.  I am honestly thinking about doing a full wipe of octoprint and starting there..but again not sure how to do that.  

To the Googles!

jdbrassell, 2020-02-04:
I went through all the steps listed here 1 by 1 and same exact issue unfortunately:
https://www.klipper3d.org/Installation.html
[klippy.log](https://github.com/KevinOConnor/klipper/files/4155088/klippy.log)

jdbrassell, 2020-02-04:
I think i've figured it out.  I read a post elsewhere of someone saying they used tty/USB0 for the connection and that's what i used instead of what I got with `ls /dev/serial/by-id/*`

I was getting this error after trying to connect:

Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2793
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2793)"
Connection closed, closing down monitor

But then ,in Octoprint, I changed my serial port to /tmp/printer and hit connect and it connected right up.  My printer LCD is even working showing a new display!!!

closed 2020-02-04 by jdbrassell