Klipper3d/klipper #2473: Fresh install on Ender 3 pro - MCU 'mcu' shutdown: Can't assign oid

gnuMoose, 2020-02-04:
I freshly installed octoprint, compiled klipper and manually flashed the firmware using progisp.  I worked through the various errors but am at a point where it's rejecting the MCU definition.  Below is the M112 output and the klippy.log file is attached.

Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Recv: ok
Closing serial port due to emergency stop M112.
Recv: ok
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline (Error: Closing serial port due to emergency stop M112.)"

[klippy.log](https://github.com/KevinOConnor/klipper/files/4154357/klippy.log)

KevinOConnor, 2020-02-06:
Your log indicates that the micro-controller reset itself during the configuration phase.  If this was just a transient error, just issue FIRMWARE_RESTART to clear the error and retry.  If the error persists, it generally indicates something is not working correctly on the micro-controller (make sure the power is stable, make sure the micro-controller really is an atmega1284p, etc.).

-Kevin

gnuMoose, 2020-02-06:
FIRMWARE_RESTART had no appreciable effect:
Send: FIRMWARE_RESTART
Recv: ok
Recv: // Klipper state: Disconnect
Send: M105
Recv: ok T:0
Send: M105
Recv: ok T:0
Send: status
Recv: // MCU 'mcu' shutdown: Can't assign oid
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: // Klipper state: Not ready
Recv: !! MCU 'mcu' shutdown: Can't assign oid
Recv: ok
Send: M105
Recv: ok T:0

The board is a a Creality Silent Mainboard V1.1.5 with TMC2208 Driver.  The only thing I can think of that I did differently is I didn't install a bootloader.  Instead I directly flashed the klipper.elf.hex to the board with progisp and a usb avr device.  It's the same process I used to flash the bltouch enabled creality firmware before trying to switch to klipper.

gnuMoose, 2020-02-06:
And, before I forget again, thank you for your support!

gnuMoose, 2020-02-06:
I went back and flashed the bootloader and went through the make flash process and things are looking MUCH better.  That will teach me to not skip steps...

Send: status
Recv: // Klipper state: Ready
Recv: ok
Send: M105
Recv: ok B:22.4 /0.0 T0:22.4 /0.0

closed 2020-02-06 by gnuMoose