Klipper3d/klipper #2474: "Move exceeds maximum extrusion" error in PrusaSlicer gcode

sohxet, 2020-02-04:
The following shows up in the terminal when I try to print from PrusaSlice:

> Send: N33 G1 Y50 E20 F1000*100
> Recv: // Move exceeds maximum extrusion (0.962mm2 vs 0.640mm2)

This is from the start gcode, the line that's supposed to preform a wipe before printing. 

Before that line, this line runs fine:

>G1 X1.0 Y-5.0 Z0.1 E1.0 F3600 

but if I change the E to 2.0 it fails with the same error and an offending value of 0.963mm^2

I've read the documentation and found multiple solutions but none that explained how 0.962mm^2 calculated, or how I can check whether an extrusion command would pass this test other than by trial and error.

sohxet, 2020-02-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4154564/klippy.log)

jakep82, 2020-02-04:
1.75mm filament is 2.405mm^3 per mm of length.  If you extrude 20mm of filament, that's 48.105mm^3.  Divide that by your Y move of 50mm and you get 0.962mm^2.   You need to use a more sane extrusion value like E10 if you're only moving 50mm, or double your Y move to 100mm.  Either will solve the issue.

sohxet, 2020-02-04:
> 1.75mm filament is 2.405mm^3 per mm of length. If you extrude 20mm of filament, that's 48.105mm^3. Divide that by your Y move of 50mm and you get 0.962mm^2. You need to use a more sane extrusion value like E10 if you're only moving 50mm, or double your Y move to 100mm. Either will solve the issue.

Amazing, thank you! The feed rate doesn't matter then?

jakep82, 2020-02-04:
Klipper won't care, but you need to make sure you stay within the limits of your hotend.  An E3D V6 can manage about 10mm^3 per second.  I should also note that a typical extrusion line of 0.45mm wide by 0.2mm high is only 0.09mm^2.  Even if you cut your extrusion value in half, you're going to be building a lot of pressure in the nozzle.  This is why I use a purge before I do a short wipe move to normalize pressure.

sohxet, 2020-02-04:
Alright, I've got the pressure advance tower printing now and I understand why it works. Thank you.

closed 2020-02-04 by sohxet