Klipper3d/klipper #2476: less then max_power: 1.0 for fan crashes klipper?

BelgarionNL, 2020-02-04:
I am trying to tame a 5020 fan for part cooling so I installed a 220uf 50v capacitor because the whining made me want to kill that printer :)

so the whining is gone but the fan is still too fast and I am trying to make it spin slower. so I used the following setting:

#max_power: 1.0
#   The maximum power (expressed as a value from 0.0 to 1.0) that the
#   pin may be set to. The value 1.0 allows the pin to be set fully
#   enabled for extended periods, while a value of 0.5 would allow the
#   pin to be enabled for no more than half the time. This setting may
#   be used to limit the total power output (over extended periods) to
#   the fan. If this value is less than 1.0 then fan speed requests
#   will be scaled between zero and max_power (for example, if
#   max_power is .9 and a fan speed of 80% is requested then the fan
#   power will be set to 72%). The default is 1.0.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4155932/klippy.log)

attached the klippy log below and also found the error? what is the correct way of slowing the fan down then?
    "shutdown value must be 0.0 or 1.0 on soft pwm")
error: shutdown value must be 0.0 or 1.0 on soft pwm

jakep82, 2020-02-04:
You just need to add a shutdown_speed value in that section, and it needs to be either 0.0 or 1.0.  The default shutdown_speed is to copy the max_power setting, but when the MCU is shut down it can't run software PWM, so it needs to be just on or off.

BelgarionNL, 2020-02-04:
oww

BelgarionNL, 2020-02-04:
perhaps you could give me one last tip. how can you set the absolute lowest speed setting on the fan?

currently tried this:

```
[heater_fan controller_fan]
# Controller fan
pin: z:ar10
#   D10 on mcu_z
max_power: 0.1
#hardware_pwm: false
kick_start_time: 0.5
heater: heater_bed
heater_temp: 50.0
#fan_speed: 1.0
cycle_time: 0.1
shutdown_speed: 0
```

but its the same with max_power: 1

jakep82, 2020-02-04:
In your first post you mentioned you're trying to slow down your part cooling fan.  What you've posted is the section for a heater fan that is activated when the bed is turned on.  If you're trying to slow down your part cooling fan, then you need to edit the [fan] section of your config.

`[fan]
pin = ar9
max_power = 0.5
kick_start_time = 0.500
`

BelgarionNL, 2020-02-04:
yay I just rebuild an upgrade in 3 days straight! I am an complete idiot! thanks!

closed 2020-02-14 by KevinOConnor