Klipper3d/klipper #2477: Move in machine coordinate system

yschroeder, 2020-02-05:
I am (ab)using Klipper for something that is not strictly a 3D printer: The cable winch robot again, if you remember.

I have the following problem:
The machine has a parking position where it parks the tool head when it is not used. I know its XYZ position and issue a `G0` to that location. When using the machine, I play gcode files with 0,0,0 at their center. To play this code at an arbitrary location I move to this location and set it as 0,0,0 with `G92`. If I want to move to the parking position afterwards, this coordinate is interpreted relative to the new 0,0,0 set with `G92` before, thus it moves to a wrong position. Is it possible to tell Klipper that it should do the move in the coordinate system of the machine, instead of the coordinate system set by `G92`?

I looked through the available commands and compared with Marlin. I think Marlin allows to use multiple coordinate systems. Can Klipper do this as well?

KevinOConnor, 2020-02-06:
Take a look at the SAVE_GCODE_STATE command in docs/G-Codes.md - that command allows one to save and subsequently restore coordinate systems.

-Kevin

KevinOConnor, 2020-02-24:
I'm going to close this issue as it looks like the original question was answered (use SAVE/RESTORE_GCODE_STATE).

-Kevin

closed 2020-02-24 by KevinOConnor