Klipper3d/klipper #2478: Build volume limits for winch kinematics

yschroeder, 2020-02-05:
I am using the winch kinematics for a very large cable robot. The winch kinematics do not allow to set a build volume limit. Cartesian kinematics can have movement limits on the axes. I'd like to be able to define the reachable coordinates (build volume) of the printer. I think this should be tuples for each axis (X, Y, Z) defining the minimum valid coordinate and the maximum valid coordinate. Any movement commands that would move outside these limits should be clipped to these limits (by applying `min()` or `max()` respectively.

I looked through the available config options but did not find the option to set overall coordinate limits.

Note: I am not looking to limit the movements of the axes (A, B, C, D in winch kinematics) but the movement in X, Y, Z coordinates of the input gcode.

KevinOConnor, 2020-02-06:
It's possible to implement the `check_move()` method in klippy/kinematics/winch.py.  (See klippy/kinematics/cartesian.py for an example of how other kinematics implement this.)

-Kevin

KevinOConnor, 2020-03-29:
I'm going to mark this as inactive.  If anyone is interested in implementing it, feel free to create a github Pull Request.

-Kevin

closed 2020-03-29 by KevinOConnor