Klipper3d/klipper #2479: Specify tool in safe_z_home

mtw3d, 2020-02-05:
For IDEX & Tool changer printers the current way that Klipper homes the Z is potentially dangerous. 

In certain situations and with certain slicers (at least Cura) it will switch to T1 at the very beginning of the gcode file. If you have a probe on T0, it will not be able to successfully probe. 

I am working on getting homing overrides set up as a workaround, but doing that breaks safe_z_home, which breaks z_hop. Not fun.

A better way to handle this would just be to allow an optional tool to be specified in safe_z_home. That would be, well, safer, and eliminate the need for people work around the limitations.

KevinOConnor, 2020-02-06:
FWIW, I look at safe_z_home as a mechanism to handle the common case where Z needs to be homed with a probe of some type.  Anything that can be done with safe_z_home can also be done with homing_override - so if a more complex homing process is needed then using homing_override is likely the next step.

-Kevin

ld3300, 2020-02-06:
Since T0 and T1 are no longer implemented by default in the current master branch you can declare them yourself.  What happens on a tool change is up to you.  For example you could put in your T1 Gcode macro:
`{% if not printer.toolhead.homed_axes %}` to clarify what to do if the tool is called but homing hasn't happened yet.

JohnEdwa, 2020-02-06:
>Anything that can be done with safe_z_home can also be done with homing_override

You can? 
What I mean is that with safe_homing G28 X and G28 Y, first Z is checked, if its unhomed or under the Z-hop height it will be lifted and then only the single axis is homed. And with G28 Z, if X or Y are unhomed the command fails, but if they are homed it will again check for the hop, then move and home Z, and then either stay or return to the starting position.

So replicating that with homing_override requires accessing the current status and position of the Z, X and Y axis, only executing the parts necessary for the individual axis without forcing all to be homed regardless of their current status and also moving back to the starting position (would that be done using SAVE and RESTORE_GCODE_POSITION?)

In any case, most likely variables and if checks would be neede,d and this would be a perfect place to add an example to the config of the more complicated Klipper features and how to use them.

KevinOConnor, 2020-02-07:
> You can?

As far as I know, yes.  It would involve g-code templates and wouldn't be particularly easy to implement every feature, but it should be possible.

-Kevin

mtw3d, 2020-02-11:
I am working on getting this going, but I believe I have found a bug in the underlying Klipper code that is causing it to fail. This is the code that I have at present:

`
# Activate the secondary extruder
[gcode_macro T1]
gcode:
    {% if not printer.toolhead.homed_axes %}
        M118 PP > Error: Not homed! {printer.toolhead.homed_axes}
    {% else %}
        M118 Homed: {printer.toolhead.homed_axes}
        {% if not printer.toolhead.extruder == "extruder1" %}
            PARK_{printer.toolhead.extruder}
            ACTIVATE_EXTRUDER EXTRUDER=extruder1
            SET_DUAL_CARRIAGE CARRIAGE=1
            SET_GCODE_OFFSET Z=-0.5
        {% endif %}
    {% endif %}
`

This works as expected if no axis is homed, however if any axis is homed, it reports that Z is homed. For example:

`Recv: // Klipper state: Ready
[...]
Send: T1
Recv: echo: PP > Error: Not homed!
Recv: ok
[...]
Send: G28 X
Recv: ok
[...]
Send: T1
Recv: echo: Homed: xz
Recv: info Macro > M106 > speed 0
Recv: info Macro > fanspeed > SET FAN to 0
Recv: info Macro > fanspeed > SET FAN realspeed to 0
Recv: // Activating extruder extruder1
Recv: ok`

Is this a bug? If not, how do I reliably determine whether the Z has actually been homed?

mtw3d, 2020-02-11:
How about this?

In addition to offering the full `[homing_override]`, would it be possible to add the ability to run a gcode macro to `safe_z_homing`? It would be far simpler and *way* less problem prone to just run the `T0` gcode macro before homing as normal then to try to recreate the entire functionality of `safe_z_homing` myself.

ld3300, 2020-02-12:
well that is certainly a bug, z should not be considered homed.  happens on my setup too.

ld3300, 2020-02-12:
hmmm, i've found the problem, now how to fix it...

ld3300, 2020-02-12:
Okay, I've submitted a fix for the z axis home problem #2500

mtw3d, 2020-02-12:
> Okay, I've submitted a fix for the z axis home problem #2500

Thanks David!

mtw3d, 2020-02-13:
So I decided to take the opportunity to treat this as a learning experience, and forked `safe_z_home.py` to add a gcode script that is run before the Z is homed. This seems to work for me:

```# Perform Z Homing at specific XY coordinates.
#
# Copyright (C) 2019 Florian Heilmann <Florian.Heilmann@gmx.net>
#
# This file may be distributed under the terms of the GNU GPLv3 license.

class MyZHoming:
    def __init__(self, config):
        self.printer = config.get_printer()
        try:
            x_pos, y_pos = config.get("home_xy_position",
                                      default=",").split(',')
            self.home_x_pos, self.home_y_pos = float(x_pos), float(y_pos)
        except:
            raise config.error("Unable to parse home_xy_position in %s" % (
                config.get_name()))

        self.z_hop = config.getfloat("z_hop", default=0.0)
        self.z_hop_speed = config.getfloat('z_hop_speed', 15., above=0.)
        self.max_z = config.getsection('stepper_z').getfloat('position_max')
        self.speed = config.getfloat('speed', 50.0, above=0.)
        self.move_to_previous = config.getboolean('move_to_previous', False)
        self.gcode = self.printer.lookup_object('gcode')
        self.gcode.register_command("G28", None)
        self.gcode.register_command("G28", self.cmd_G28)
        gcode_macro = self.printer.try_load_module(config, 'gcode_macro')
        self.template = gcode_macro.load_template(config, 'gcode')

        if config.has_section("homing_override"):
            raise config.error("homing_override and safe_z_homing cannot"
                               +" be used simultaneously")

    def cmd_G28(self, params):
        toolhead = self.printer.lookup_object('toolhead')

        # Perform Z Hop if necessary
        if self.z_hop != 0.0:
            pos = toolhead.get_position()
            curtime = self.printer.get_reactor().monotonic()
            kin_status = toolhead.get_kinematics().get_status(curtime)
            # Check if Z axis is homed or has a known position
            if 'z' in kin_status['homed_axes']:
                # Check if the zhop would exceed the printer limits
                if pos[2] + self.z_hop > self.max_z:
                    self.gcode.respond_info(
                        "No zhop performed, target Z out of bounds: " +
                        str(pos[2] + self.z_hop)
                    )
                elif pos[2] < self.z_hop:
                    self._perform_z_hop(pos)
            else:
                self._perform_z_hop(pos)
                toolhead.get_kinematics().reset_limits(homing_axes=[2])

        # Determine which axes we need to home
        if not any([axis in params.keys() for axis in ['X', 'Y', 'Z']]):
            need_x, need_y, need_z = [True] * 3
        else:
            need_x, need_y, need_z = tuple(axis in params
                                           for axis in ['X', 'Y', 'Z'])

        # Home XY axes if necessary
        new_params = {}
        if need_x:
            new_params['X'] = '0'
        if need_y:
            new_params['Y'] = '0'
        if new_params:
            self.gcode.cmd_G28(new_params)
        # Home Z axis if necessary
        if need_z:
            try:
                self.in_script = True
                self.template.run_gcode_from_command()
            finally:
                self.in_script = False

            # Move to safe XY homing position
            pos = toolhead.get_position()
            prev_x = pos[0]
            prev_y = pos[1]
            pos[0] = self.home_x_pos
            pos[1] = self.home_y_pos
            toolhead.move(pos, self.speed)
            self.gcode.reset_last_position()
            # Home Z
            self.gcode.cmd_G28({'Z': '0'})
            # Perform Z Hop again for pressure-based probes
            pos = toolhead.get_position()
            if self.z_hop:
                pos[2] = self.z_hop
                toolhead.move(pos, self.z_hop_speed)
            # Move XY back to previous positions
            if self.move_to_previous:
                pos[0] = prev_x
                pos[1] = prev_y
                toolhead.move(pos, self.speed)
            self.gcode.reset_last_position()

    def _perform_z_hop(self, pos):
        toolhead = self.printer.lookup_object('toolhead')
        # Perform the Z-Hop
        toolhead.set_position(pos, homing_axes=[2])
        pos[2] = pos[2] + self.z_hop
        toolhead.move(pos, self.z_hop_speed)
        self.gcode.reset_last_position()

def load_config(config):
    return MyZHoming(config)
```

Personally, I think this should be added to `safe_z_home` since it is far simpler and more reliable to simply add `T0` to your `safe_z_home` config then it is to try to recreate the functionality within `homing_override`. But before I submit a pull request, I wanted to post the code here to see if others agree with me.

vladbabii, 2020-02-19:
For what is worth, i have only probe on T1, T0 is no tool (to make dropping any current tool easier), and here is my gcode:

```
[homing_override]
gcode:

  RESPOND PREFIX="info" MSG="Homing > Resetting gcode offset"
  SET_GCODE_OFFSET X=0 Y=0 Z=0

  # go 2 cm down
  SET_KINEMATIC_POSITION Z=0
  G91 
  G0 Z20
  G90

  RESPOND PREFIX="info" MSG="Homing > Y"
  G28 Y
  G91
  G0 Y-10
  G90

  RESPOND PREFIX="info" MSG="Homing > X"
  G28 X
  G91
  G0 X5
  G90

  RESPOND PREFIX="info" MSG="Moving away from X Y"
  G91
  G0 Y-15 X15
  G90

  SET_GCODE_VARIABLE MACRO=homing_status VARIABLE=homed VALUE=1.0

  ## is this needed ?
  RESPOND PREFIX="info" MSG="Homing fake Z"
  SET_KINEMATIC_POSITION Z=0

  ## home toolchanger
  {% if printer["gcode_macro toolch_config"].homed|int == 0 %} 
    TOOLCH_HOME
  {% endif %}

  # # minimum safe z heigh - 4 cm ??
  RESPOND PREFIX="info" MSG="Homing > lifting up 4 cm"
  G91 
  G0 Z40
  G90
  SET_KINEMATIC_POSITION Z=0

  # drop current tool if mounted
  # TODO

  # # pickup probe
  # # TODO: detect probe by iterating through tools
  T1

  # # absolute move to bed center
  RESPOND PREFIX="info" MSG="Homing > Moving to bed center"
  G90
  #G0 X220 Y100 F5000
  G0 X225 Y140

  # # probe z
  # RESPOND PREFIX="info" MSG="Homing > Probing Z"
  G28 Z0

  # # lower bed
  G91
  G0 Z40
  G90

  # # drop tools
  T0 

  RESPOND PREFIX="info" MSG="Homing > Finished!"
```

I needed the SET_KINEMATIC_POSITION Z=0 command to work well with tool offsets - so if i have to drop/pickup a tool while homing i have no errors about z not being homed. The important part is that i move the bed down 4 cm to avoid crashes because i have no idea what Z height i am right now (on first home) and my tools have at maximum a couple of mm of offset on z - 40 mm makes sure there's zero risk of anything being hit/crashed

KevinOConnor, 2020-03-05:
I'm going to close this as it looks like a workaround was found for the original report.

-Kevin

closed 2020-03-05 by KevinOConnor