Klipper3d/klipper #2482: Is it possible to retrieve the hotend and bed PWM duty value?

dewi-ny-je, 2020-02-06:
When I was using Repetier with Marlin I was able to see a graph of the PWM values for the heaters. This was an indication of how close I was to the power limits.

Is it possible to obtain the same using Klipper? It doesn't need to be continuous, extracting it from the log at the end of a test is enough, just like the Octoklipper plugin does to plot MCU occupation and bandwidth.

Arksine, 2020-02-06:
Yes, they are in the log.  You can use graphstats.py to build a load graph for each heater using the -t option, ie:
```
~/klipper/scripts/graphstats.py /tmp/klippy.log  -o tempgraph.png -t extruder
```

The result will be something like this:
[image]

Note that graphstats.py depends on matplotlib, which can take 30 minutes+ to install on a Raspberry Pi.  If you have used Octoklipper to generate load graphs then you likely already have it.

KevinOConnor, 2020-02-07:
FYI, on the RPi you should be able to just install the OS version of matplotlib which should only take a minute or so - see: https://www.klipper3d.org/Debugging.html#generating-load-graphs

-Kevin

Arksine, 2020-02-07:
Thanks Kevin, all this time and I didn't know that there was a package for it.

dewi-ny-je, 2020-02-07:
I will try to implement it in Octoklipper

KevinOConnor, 2020-02-24:
Okay, I'm going to close this as it looks like the original question was answered (yes, the values are in the log).

-Kevin

closed 2020-02-24 by KevinOConnor