Klipper3d/klipper #2484: Nested gcode commands not working in current release

w1ebr, 2020-02-07:
Up until today, when I installed the latest release of Klipper, I have had no problem with including a gcode macro inside of a gcode macro. In the current release it says "MALFORMED COMMAND" and the nested gcode macro name "SET_CARRIAGE"

Several gcode macros include a gcode macro called "SET_CARRIAGE"

[klippy.log](https://github.com/KevinOConnor/klipper/files/4168880/klippy.log)

KevinOConnor, 2020-02-07:
It doesn't like the ';' - there is no need to add semicolons in the gcode macros.  (I recommend removing all semicolons from from your macros.)  Klipper could have done better here (a semicolon generally starts a comment in g-code).  I'm not sure why it worked previously.

-Kevin

w1ebr, 2020-02-07:
Great! I'll try it now!

w1ebr, 2020-02-07:
Thank you, Kevin!

closed 2020-02-07 by w1ebr