Klipper3d/klipper #2485: Disconnects, 'no next step' - multi-mcu, too much for RPi3B v1.2?

daninfuchs, 2020-02-07:
Hi there! I'm prototyping a printer, and I've run into a snag. Should be an easy one for this particular question.

I've got a delta, with three extruders. Raspberry Pi 3b v1.2 as host. Six motors, so obviously six stepper drivers. Herein lies my rub; I'm trying to do simple step distance tuning, so I'm executing FORCE_MOVE on the A-tower motor, 100mm at 50mm/sec, a fairly safe move speed. Intermittently -- sometimes mid-move, sometimes immediately after starting -- it stalls with no next step.

Final details (I believe?) relevant to the question;
MCU0 (mcu); Mega2560 + RAMPS1.4 clone
 - Three steppers (Tower A/B/C)
 - Extruder Shared heater/sensor/fan
 - Bed heater (Aux-1 to external FET)
 - The MCU doing the move, which is also shutting down first

MCU1 (grbl); Duemilanove + GRBL CNC Shield
 - Three steppers (Extruder 0/1/2)
 - Seems to shutdown as a response to the first mcu failure?

(I've got more equipment planned but I wanted this much operational first.)

Now down to the bones of the query -- am I asking too much of the RPi 3B v1.2 itself? If so, should I acquire an RPi3B+? Would that also be not-good-enough?

Or, most importantly - is the RPi the place I should be looking at all? I realize "it could be anything" but honestly I'm just looking for someone to say "the RPi3Bv1.2 should [not] run dual MCU reliably" so I know which problem I'm actually chasing. Or maybe I've overlooked something really dumb.

Thanks for your attention!

daninfuchs, 2020-02-07:
(whoops)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4169568/klippy.log)

KevinOConnor, 2020-02-07:
The problem originates from the FORCE_MOVE command.  That command does not (by default) implement acceleration, and moving at a fast speed triggers a safety check programmed into the micro-controller.  The solution is to not use FORCE_MOVE, or to specify an ACCEL parameter to FORCE_MOVE (see docs/G-Codes.md for the details).

Separately, there's a lot of processing power in an RPi3.  Klipper typically uses no more than about 10% of one core of an RPi3.

-Kevin

daninfuchs, 2020-02-07:
Fantastic, thank you. That's what I wanted to hear.

I understood the risks and instabilities with using the command re: machine state/consistency, acceleration, and so on -- though at 50mm/s on a motor connected to only a belt (no carriage, calibration phase) there's no real need for acceleration so I discarded the acceleration for ease of typing.

I should note, the G-Codes page on the Klipper wiki makes no mention of potentially firing an MCU error that seems entirely unrelated on using this command - I understand that its use is discouraged but perhaps a note there could be added?

Regarding the RPi3, I understand that it has a very beefy engine under the hood, but my assumption of the possible issue was more one of precise timing than of its ability to do the work required. I know it cannot be used, for example, directly controlling stepper motor drivers via GPIO because its timing at that level is not good enough -- I wasn't sure if this was related, as I'm uncertain of the timing requirements within the Klippy/klipper datapath. Very pleased to hear it isn't a problem!

I suppose on a final query; other than FORCE_MOVE is there a means you would suggest to calibrate step distance on a Delta tower with its carriage disconnected (and no probe) or should merely specifying a sane acceleration value and moving at a safe speed be good enough? I'll have a probe for final pass, arm/radius calibration, etc, but for initial moves I'd like it geometrically *closer* and I'm curious what the internal word is there..

Thanks Kevin!

closed 2020-02-07 by daninfuchs