Klipper3d/klipper #2488: stm32f103 Clock Referense, Internal clock, no work.

Grey-stud, 2020-02-07:
Code has an error, file
https://github.com/KevinOConnor/klipper/blob/master/src/stm32/stm32f1.c
https://github.com/KevinOConnor/klipper/blob/b3a180ea7723eb11cde757b25dc7f973765f0cf4/src/stm32/stm32f1.c#L138

div2 value = (72000000/8000000) * 2=18
cfgr = ((0 << RCC_CFGR_PLLSRC_Pos)
| ((div2 - 2) << RCC_CFGR_PLLMULL_Pos));
Bits RCC_CFGR_PLLMULL = 16 split
ST RM0008 Reference manual, page 134, PLLMUL only 4 bits
Instead of 72MHz, 8MHz only

Grey-stud, 2020-02-07:
From an internal source, the maximum core frequency can only be 64MHz
[image]

KevinOConnor, 2020-02-14:
Thanks.  Should be fixed now (commit b5c967ac).

-Kevin

closed 2020-02-24 by KevinOConnor