Klipper3d/klipper #2489: Removal of T commands!?

radensb, 2020-02-07:
Quick Question. I am planning on upgrading klipper from V0.7 (installed in ~May 2019) and I just read this in the Change log:

> 20191210: The builtin T0, T1, T2, … commands have been removed. The extruder activate_gcode and deactivate_gcode config options have been removed. If these commands (and scripts) are needed then define individual [gcode_macro T0] style macros that call the ACTIVATE_EXTRUDER command.

I have a dual extruder, so this seems like a pretty important change. The solution with the gcode macro seems simple enough - but what was the reasoning for removing native support for this command? It seems like such a fundamental command that will be in any gcode file created for a multi-extruder printer from every slicer tool in existence. Is there some advantage in removing it to have a gcode macro replace it? Just curious and want to make sure I understand this correctly before upgrading.
Cheers!

--- I know, I know... there is no Klippy log.

KevinOConnor, 2020-02-07:
FYI, this was discussed a bit in PR #2207 .

-Kevin

radensb, 2020-02-07:
Thanks Kevin. Not sure how I missed that discussion, but it is a good read. If I follow the upgrade instructions by doing:
`git pull`
`~/klipper/scripts/install-octopi.sh`

Will I be installing the current master branch head revision, or the tagged v0.8 release? I see that the T command commit was after the v0.8 release.

I appreciate the clarification!

ld3300, 2020-02-08:
To add my 2 cents, implementing the commands myself made the filament changing I needed to do so much easier than the way they were implemented previously.

radensb, 2020-02-08:
As an alternative to START and END extruder blocks, this is an interesting idea. I do like the idea of the Klipper configuration having more control over stuff. Closing this as the Kevin provided the reference to the discussion and I answered my own question regarding upgrade branch. It does use the head of the master branch as confirmed by the commit tag ID.
Thanks everyone.

closed 2020-02-08 by radensb