Klipper3d/klipper #2490: Bed mesh output full table is inverted relative probe data

GarthSnyder, 2020-02-07:
In a BED_MESH_OUTPUT report, is there a considered reason why the raw data table and full (interpolated) data table are in inverse order, or is this just happenstance? If the latter, it would make analyzing this data more convenient and less confusing if they used the same ordering scheme.

The left-to-right order within each line appears to be the same, but the lines of the full table are reversed. For example, in the following example, note that 0.051047 is the first entry of the last line of the raw data table and the first entry of the first line of the full table. Its raw-data neighbor 0.071058 is the second entry of the last line of the raw data table and the third entry of the first line of the full table. (Not the second entry, because that's an interpolated value in this case.)

The raw data table appears as it does in printer.cfg, so I would suggest reversing the full table.

```
Send: bed_mesh_output
Recv: // Mesh Leveling Probed Z positions:
Recv: // -0.057346 -0.090697 -0.098202 -0.081526 -0.051509 -0.073188
Recv: // -0.029831 -0.068185 -0.085695 -0.086528 -0.057346 -0.077357
Recv: // -0.017324 -0.050675 -0.049008 -0.037335 -0.013155 -0.025662
Recv: // -0.054011 -0.043171 -0.021493 0.014360 0.071058 0.088568
Recv: // -0.079024 -0.045673 -0.023994 0.035205 0.098573 0.135260
Recv: // 0.051047 0.071058 0.081064 0.127756 0.188623 0.211969
Recv: Mesh X,Y: 11,11
Recv: Search Height: 5
Recv: Mesh Average: -0.01
Recv: Mesh Range: min=-0.0882 max=0.2220
Recv: Interpolation Algorithm: lagrange
Recv: Measured points:
Recv:   0.051047  0.069019  0.071058  0.072303  0.081064  0.100078  0.127756  0.159434  0.188623  0.208263  0.211969
Recv:   -0.045219  -0.018472  -0.018735  -0.017759  -0.003686  0.023678  0.057584  0.089056  0.111617  0.126009  0.144910
Recv:   -0.079024  -0.052265  -0.045673  -0.039380  -0.023994  0.002062  0.035205  0.069351  0.098573  0.119757  0.135260
Recv:   -0.075152  -0.057636  -0.046764  -0.036453  -0.022167  -0.001240  0.026792  0.059866  0.093059  0.118265  0.123864
Recv:   -0.054011  -0.051782  -0.043171  -0.032996  -0.021493  -0.006498  0.014360  0.041471  0.071058  0.092984  0.088568
Recv:   -0.031241  -0.045220  -0.044099  -0.038135  -0.030798  -0.021431  -0.007908  0.010708  0.031508  0.045076  0.032831
Recv:   -0.017324  -0.043181  -0.050675  -0.051082  -0.049008  -0.044862  -0.037335  -0.025874  -0.013155  -0.007563  -0.025662
Recv:   -0.017190  -0.047011  -0.059831  -0.066121  -0.069706  -0.070363  -0.066423  -0.057374  -0.046458  -0.043276  -0.066384
Recv:   -0.029831  -0.055567  -0.068185  -0.077585  -0.085695  -0.089909  -0.086528  -0.074197  -0.057346  -0.049628  -0.077357
Recv:   -0.047904  -0.066616  -0.075927  -0.084840  -0.093554  -0.097529  -0.091534  -0.073714  -0.049643  -0.036381  -0.066534
Recv:   -0.057346  -0.078229  -0.090697  -0.097264  -0.098202  -0.092952  -0.081526  -0.065918  -0.051509  -0.048474  -0.073188
```

Arksine, 2020-02-08:
Indeed, the interpolated values in BED_MESH_OUTPUT are inverted as if you were looking down at the bed, with the origin in the bottom left corner.  It was done this way to retain compatibility with other firmware.

KevinOConnor, 2020-03-05:
I'm going to close this as it looks like the original question was answered (the software is designed to use that orientation for the output).

-Kevin

closed 2020-03-05 by KevinOConnor

GarthSnyder, 2020-03-05:
Yes, thanks.