Klipper3d/klipper #2491: [FEATURE REQUEST] tmc 2209-5160 coolstep

Coffee0297, 2020-02-08:
https://github.com/MarlinFirmware/Marlin/pull/16790

Marlin just got a pull for coolstep, dont look too bad to implement in klipper. it wuld be abselutly perfect for 3d printers, wich are what klipper focus on.

-Tonny

FHeilmann, 2020-02-10:
This is already possible using SET_TMC_FIELD, just pass the appropriate field names and the values and you'll be able to set up coolstep:
e.g.

`SET_TMC_FIELD STEPPER=stepper_y FIELD=seup VALUE=2`

the tmc2660 and tmc5160 also allow you to set these values via the config.

Keep in mind that coolstep requires a tuned stallguard value, and that the stallguard tuning is very susceptible to things like ambient temperature, movement speed and stepper motor temperature, so getting one tune for all situations is quite tricky.

-Florian

Coffee0297, 2020-02-10:
great then it shuld not be hard to make a setting that says coolstep=true
so we shuld not do it at each controller restart
-Tonny

thisiskeithb, 2020-02-11:
CoolStep [was removed](https://github.com/MarlinFirmware/Marlin/commit/18a7276909a3e2aa1761ba0c46577a8f6e7baedb) almost as quickly as it was added to Marlin, so YMMV.

Fabi0San, 2020-02-11:
It needs to be cautiously tuned to your steppers and moving mass before being enabled.
The settings merged to Marlin were tuned to my setup :( I tried to warn https://github.com/MarlinFirmware/Marlin/pull/16790#discussion_r376750617 but that PR went in quite quickly.

KevinOConnor, 2020-02-24:
FYI, I'm not aware of a way to automatically configure "coolstep".  To the best of my knowledge it requires external probing hardware - there's info on doing the necessary calibration in the trinamic documentation.  Should someone go through those steps, it is possible to configure Klipper with the resulting values.

-Kevin

Fabi0San, 2020-02-24:
No one setting will fit every machine, so shipping Klipper with one-size-fits-all setting is unfortunately not possible.

It is possible though to create a procedure to automatically gather the actual SG_RESULT for a given machine at given speed ranges, and use that to tune coolstep, and stallguard. No special hardware should be required. It could be similar to the home stepper phase calibration.

Cheers,
   Fabio.

KevinOConnor, 2020-02-24:
> It is possible though to create a procedure to automatically gather the actual SG_RESULT for a given machine at given speed ranges, and use that to tune coolstep, and stallguard. No special hardware should be required. It could be similar to the home stepper phase calibration.

Okay, thanks for the info.  If a developer is interested in writing such a tuning tool that would be great.

Cheers,
-Kevin

Fabi0San, 2020-02-24:
I have to say I'm very tempted :), I just heard about Klipper because of this thread and I'm really impressed by what you've done. 
I've been chasing mathematical purity on my fast(500 mm/s) delta for a few years now, after a few months modeling everything on Mathematica it was clear to me that chasing the real time acceleration computation with a complex delta geometry(diff arm lenghts, different belt tension, etc) would require more HP than even the 32bit boads could give, and it seemed silly given all the computational power available at the host. I even dabbed at creating a protocol to efficiently transferring step timings.

All that to say I really get what you are going for, I'll learn a bit more about Klipper and may take on this and other opportunities to make it better :)

Cheers,
    Fabio.

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it looks like the conversation has concluded.  If a developer is interested in adding support feel free to open a pull request with the changes.

-Kevin

closed 2020-03-21 by KevinOConnor