Klipper3d/klipper #2492: Invalid M204 Command ...

radensb, 2020-02-09:
I noticed (about 8 hrs into a 12 hr print) that the acceleration commands generated in Cura 4.3 using the RepRap flavor GCODE were not being recognized properly. I thought RepRap was the prefered flavor to use with Klipper. Is this not correct? Klipper docs state to refer to the RepRap documentation for GCODE info:

> For further details on the above commands see the RepRap G-Code documentation.

In my GCODE file, I have:
`M204 P1500`

The docs say:
> - Set acceleration: M204 S<value> OR M204 P<value> T<value>
> -- Note: If S is not specified and both P and T are specified, then the acceleration is set to the minimum of P and T.

When I change the GCODE to Marlin, I get:
`M204 S1500`

I seem to recall that using Marlin flavor with Klipper had compatability issues that were corrected with RepRap flavor, but now the opposite seems to be the case. Can anyone comment on the proper flavor to use with v0.8 with any pitfalls to watch out for?

Thanks!
[klippy.log](https://github.com/KevinOConnor/klipper/files/4176696/klippy.log)

theopensourcerer, 2020-02-09:
https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md#set-the-g-code-flavor-to-marlin

Hywelmartin, 2020-02-10:
This must be a slicer specific... thing.. Prusaslicer (Slic3er) Reprap/Sprinter flavor produces M204 Sxxxxx but Marlin flavor M204 Pxx Txx Rxxx..
I believe that Klippers interpretation of Marlins M204 is a bit ruff.. the lowest value of Pxx and Txx for all movements...
I use  Reprap/Sprinter Flavor in Prusaslicer and it works good.. 

After some quick investigation.....
Smoothie might be the one to chose... in all slicers.... there is no difference in the gcode between  Reprap/Sprinter and Smoothie Flavor in Prusaslicer (I assume the same for Slic3r)..

radensb, 2020-02-10:
Thanks for the replies. Marlin/Smoothie seems like the way to go. I will have to slice and compare some smoothie files as I have never used it before. I now recall more about why I thought I had to move away from Marlin with Klipper .... I am coming over from stock Lulzbot TAZ6 firmware and their version of CURA. I recall that the Lulzbot version of CURA generated M104 and M109 commands that used the 'R' parameter, which Klipper does not support. I solved this by switching to RepRap, which I believed was the better choice as it was referenced in the Gcodes.md and solved the "R" parameter issue. Now that I am using Ultimaker CURA (v4.3), I see that I can slice with Marlin flavor selected and get M204 commands with the single 'S' parameter, and M104/M106 commands with the 'S' parameter - which Klipper should be just fine with.

Maybe the Marlin Gcode referance page should be linked to instead of RepRap?

Also, is there a tool that could be used to use the Klipper engine to parse a Gcode file and report "unknown commands" or syntax before running a print? It would be nice to use something like this to check a gcode file to ensure that it will run as intended.

GarthSnyder, 2020-02-25:
I was just looking through Klipper's M204 code and I noticed that _both_ P and T must be present in order for Klipper to do anything with either one of them. A form such as M204 P1500 raises an "Invalid M204 command" exception, which explains the OP's initial problem.

Is this in fact the intended behavior?

KevinOConnor, 2020-02-25:
> Is this in fact the intended behavior?

Yes - see docs/G-Codes.md.

-Kevin

KevinOConnor, 2020-03-21:
I'm going to close this ticket as it appears the original question was answered.

-Kevin

closed 2020-03-21 by KevinOConnor

robustini, 2021-08-31:
@KevinOConnor, documentation is unclear.
"If S is not specified and both P and T are specified, then the acceleration is set to the minimum of P and T. If only one of P or T is specified, the command has no effect"... what does "at a minimum" mean?
What is stated in the cfg?
So there is no possibility to change P and T distinctly with M204?

pandel, 2021-08-31:
@robustini You can! Just define your own M204. See this as reference: https://github.com/Desuuuu/klipper-macros/blob/master/macros/M204.cfg