Klipper3d/klipper #2494: unable to restart klipper after power outage - Octoprint complains it has no permissions to read /tmp/printer

joonadec, 2020-02-10:
I had a power outage which caused my host (an Orange Pi PC running Armbian) to crash.

Upon restarting the Orange Pi, klipper would not restart, and Octoprint gave me an error saying it had no permissions to read /tmp/printer.

There was no klippy.log, and 'service klipper status' showed that klipper was not running.

After a bit of googling, I was able to fix this by renaming /tmp/printer -> /tmp/printer.old and restarting klipper.

Not really a showstopper, but shouldn't klipper check for stale links to /tmp/printer and clean it up? Or spit out an error message? Or at least produce a logfile?

joonadec, 2020-02-10:
No log file to attach...

KevinOConnor, 2020-02-14:
Okay, but unfortunately there isn't enough information for us to take any particular action.  As far as we know, the software works as intended.

-Kevin

KevinOConnor, 2020-03-05:
I'm going to close this as it appears there isn't much we can do without further information.

-Kevin

closed 2020-03-05 by KevinOConnor