Klipper3d/klipper #2495: New idea

yoc1983, 2020-02-11:
Hi,
i wasn't sure where to post this
it's not and issue but an idea to help the config process
a config UI
very simple one that will hide all the #commented info 
making it easily readable for config and showing the info when needed
also check/uncheck comments
i would do it myself but i am not a developer
anyway love klipper and wanted to suggest

KevinOConnor, 2020-02-18:
Certainly sounds useful.  An interested developer would need to undertake it though.

-Kevin

KevinOConnor, 2020-03-05:
I'm going to close this as it looks like the conversation has concluded.  There is certainly interest in adding an http interface to Klipper - so hopefully we'll see something like this in the future.

-Kevin

closed 2020-03-05 by KevinOConnor