Klipper3d/klipper #2496: absolute/relative coordinates G90, G91 it seems not change between them

tylez, 2020-02-11:
Hi,

I'm experienced a strange behaviour when I want to change between absolute and relative mode. When I started using G0 commands the printer responds correctly but when I switch to relative mode it seems klipper does not change and the next command does not cause effect.

G0 X10
M114 -> x:10
M105
G0 X25
M114 -> x: 25
M105
G0 X30
M114 -> X:30
M105
G91 X30 -> X:30 ----> this movement it seems does not execute
M114: 
G91 X10
M114  -> x:40 but if we repeat it works

I'm not sure if the commands G90 and G91 can be use with other commands together just like one command line:

G91 G0 X10  or if it needs a separate command
G91
G0 X10

[klippy.log](https://github.com/KevinOConnor/klipper/files/4185361/klippy.log)

Alejandro Guerra <tylez87@gmail.com>

ld3300, 2020-02-11:
G90 and G91 don't take parameters, so your second thought is right, the correct syntax is
```
G91  # puts us in relative mode
G0 X10  # moves X to current position plus 10
```
Any additional moves will continue to be relative until you send G90 to return to absolute mode

tylez, 2020-02-11:
Ok, thanks a lot @ld3300, I was just trying to execute at the same time. I'll change my flow to send commands.

closed 2020-02-11 by tylez