Klipper3d/klipper #2498: Max31856: Cold Junction High Fault & Cold Junction Range Fault

albmargar30, 2020-02-11:
Hello, I am trying to connect a type K thermocouple to a board that carries an STM32F103 and a Max31856 from MCU to read the thermocouple. I don't know why it fails because the LED on the FAULT board always comes on and I might be skipping a parameter in the configuration file.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4186571/klippy.log)

ld3300, 2020-02-11:
looks like it can't read the UART pin of tmc2208 stepper_x.  Try commenting it out temporarily and see if it starts up successfully.

KevinOConnor, 2020-03-05:
FYI, there were a couple of recent issues found in the stm32f103 code that were recently fixed.  Best would be to retry with the latest code.

-Kevin

KevinOConnor, 2020-03-21:
I'm going to close this ticket as the original issue may be fixed with the latest code.

-Kevin

closed 2020-03-21 by KevinOConnor

KevinOConnor, 2020-11-23:
Unfortunately, the Klipper initialization sequence for the MAX31856 board was defective (see #3567). It should now be fixed (commit 665ec5e).

-Kevin