Klipper3d/klipper #2499: Lost USB Communication Errors - Webcam not connected

bwittig, 2020-02-11:
I was also having this issue. Verified my power supply (no under voltage errors in log), USB cables are good quality and short.
It turns out that Octoprint was trying to start mjpg_streamer at the same time Klipper was freezing/aborting.
I connected my C270 webcam and all is working now.

KevinOConnor, 2020-02-18:
I'm not sure what is being reported here.  It sounds like you've found the solution though.  I think we'll need the log of the original issue to take away anything further though.

-Kevin

bwittig, 2020-02-18:
I just wanted to make you aware of what I found to solve the issue since it
seems to be a faq and I didn't find my solution documented anywhere.
It's not really an Issue, but that seemed to be the best way to let you
know.

closed 2020-02-18 by klipper-gitissuebot