Klipper3d/klipper #2501: Uneven first layer with bltouch

yoc1983, 2020-02-12:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4193424/klippy.zip)

I added an original bltouch V3.0
and it's probing and all is working
i can even see the result in bed visualizer plugin

but when i try to print a large square
i get uneven results,
one side of the bed is perfect
the other is too close to nozzle
i can tune the z offset while printing
and fix this side but then the other one is ruined

i think there is no compensation
cause i can't see movement in z-axis
i checked for backlash and it's not noticeable
~0.02mm movements so i guess it's good
any advice?

yoc1983, 2020-02-12:
ok, i noticed in the log after probing the bed it's written 
'bed_mesh fade complete: Current Z: 10.0000 fade_target: 0.0500 '
i am printing only one layer for testing so it's about 0.1mm 
fade_target on default so i guess he averaged the height and cancelled the leveling

so next i tried to bypass by setting 'fade_target: 10', but it was out of range
and gave an error in terminal stopping the print right after probing the bed, my bad
Recv: // bed_mesh: ERROR, fade_target lies outside of mesh z range
Recv: // min: -0.1075, max: 0.2375, fade_target: 10.0000

so now i uploaded a 20mm high model and uncomment the fade_target 
but still somehow got an error
bed_mesh: ERROR, fade_target lies outside of mesh z range
min: -0.1075, max: 0.2375, fade_target: 10.0000

i don't understand the difference between fade_target and fade_end also..

jakep82, 2020-02-12:
This is explained in the bed mesh document.  You should not configure fade_target (default is 0) unless you have a specific reason.  What you should be configuring are fade_start and fade_end.

https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md#mesh-fade

Arksine, 2020-02-13:
I recommend removing the fade options to disable fade completely until you are confident that bed leveling is working correctly. 

Generally speaking leveling issues are typically related to either the probe or the frame.   I'd recommend following the procedures below to check the probe accuracy and location bias.

https://github.com/KevinOConnor/klipper/blob/0c247e55e6ec62664d28f1274c553919ce870983/docs/Probe_Calibrate.md#repeatability-check

yoc1983, 2020-02-13:
thanks guys,
@jakep82 ,@Arksine 
as noted on my 3rd try i uncommented fade_target and still the print came out bad
 i did a repeatability check and it came out good 
[image]
i didn't check location bias but i will do now

Arksine, 2020-02-13:
@yoc1983 I would need to see a log of your 3rd try to diagnose that issue.  If you uncommented fade_target no error should have been raised for it being outside of the detected range.

yoc1983, 2020-02-13:
@Arksine 
i did the location bias and it was a little bit off from required(0.025)
range was between max and min is 0.038 .
but i don't understand how the requirement is so tight since check is partially manual (with paper)

any i also rechecked everything and also found x,y offset was 1-2 mm off
i know i can't trust those creality CAD's when designing the bltouch bracket
but more then 1mm is a lot. 

so i re-run a test and it work!!!!!
yiiiiiii
thanks for the help guys
we can call this one closed

johngianni, 2020-02-18:
The solution to this for me was to home and create the mesh with the bed at printing temperature.

KevinOConnor, 2020-02-24:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-02-24 by KevinOConnor